cartographer-project/cartographer_ros #268: Problem getting simulated SLAM up and running

SirVer, 2017-02-23:
Thread on the mailinglist:

 https://groups.google.com/d/msg/google-cartographer/ZQUwvfn7Nbw/cFYXhbYMBAAJ

SirVer, 2017-05-04:
/cc @mohamadfarhat

Sorry for taking so long with this.

The [provided bag](https://www.dropbox.com/sh/ijzdht7cq9nd0i2/AADBtaSE9jmHgrvSPFEjoea4a?dl=0) contained a super noisy IMU with very large accelerations and the `/tf` topic contained various improperly timed tf messages.

My changes are in [this branch](https://github.com/googlecartographer/cartographer_ros/compare/master...SirVer:user/mohamadfarhat). I only filtered some of the bad `/tf` messages to reduce noise on the terminal. I then ran `roslaunch cartographer_ros 3d_google_cartographer.launch bag_filename:=$HOME/Downloads/sensoren_daten.bag`, waited for the completion of the rosbag, then ran  `rosservice call /finish_trajectory map`. This gave me the map below.

It is not perfect, but that is not surprising given the extreme IMU noise, but in general it works.

[image]

closed 2017-05-04 by SirVer

mohamadfarhat, 2017-05-07:
@SirVer  Thanks you. You helped me a lot. I have another question. I read the Configuration and the ROS API.  Why IMU is required for using 3D Cartographer and optional for 2D? Why it is not Possible to use 3D Cartographer  without IMU? Thanks

SirVer, 2017-05-08:
I added a FAQ entry for this question: https://google-cartographer-ros.readthedocs.io/en/latest/faq.html