cartographer-project/cartographer_ros #269: How do the two single-line lasers generate 3D pointcloud with cartograher

alphakang, 2017-02-23:
I try to use two single-line lasers(Hokuyo UTM-30LX) and an IMU to generate indoor 3D pointcloud. One mounted horizontally and one vertically. I put them on a wheel barrow to collect a set of data, and recorded a bag, the bag info shows as below:
[image]

I used the horizontal laser to test the 2D map generation, it works well. But when I tried to use them for 3D, I found there are no PointCloud2 topics, so I tried to use the [laser_geometry ](https://github.com/ros-perception/laser_geometry)to convert laser scan data into pointclouds. But I found all the matched pointclouds are around the map origin and don't change with the sensors position.

So I doubt is my sensor integration solution wrong. I checked your 3D sample data which used two Velodyne VLP-16 lasers, it's diffrent with my single-line laser. So my questions are:

1. Does Cartographer support two single-line lasers to generate 3D pointcloud? Should I change to use axially rotated scanner or a 3D laser like Velodyne VLP-16?
2. If support, why I can't get the right pointclouds using my solution?

Thanks

SirVer, 2017-02-23:
You cannot do 3D SLAM with single line laser, since there is no overlap between scans in the orthogonal movement direction.

You can however use the horizontal for 2D SLAM and visualize the other one using the asset writer to get you point clouds and x-rays. It that what you want?

alphakang, 2017-02-24:
@SirVer Yes, thank you so much for your explanation. It's very helpful and saves a lot of time for me. I will try the asset writer.

alphakang, 2017-02-28:
@SirVer Can you give me some sample codes for using Asset writer? Thanks.

SirVer, 2017-03-01:
@alphakang See [this post from wohe](https://github.com/googlecartographer/cartographer/issues/212#issuecomment-283304003).

closed 2017-03-01 by SirVer