cartographer-project/cartographer_ros #270: Check failed: proto.range_min() >= 0.f (-2.35183 vs. 0)

kintzhao, 2017-02-24:
I test it using ros command to test the latest code and meet following error.    
>  roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=/home/yhzhao/rosBagData/cartographer/cartographer_paper_deutsches_museum.bag

---------------------------------------------------------------------
yhzhao@X550JD:~/catkin_ws$ roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=/home/yhzhao/rosBagData/cartographer/cartographer_paper_deutsches_museum.bag
... logging to /home/yhzhao/.ros/log/81843de6-fa70-11e6-8456-08626614ffb2/roslaunch-X550JD-14662.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://X550JD:57262/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    playbag (rosbag/play)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [14674]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 81843de6-fa70-11e6-8456-08626614ffb2
process[rosout-1]: started with pid [14687]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [14690]
process[cartographer_node-3]: started with pid [14699]
process[rviz-4]: started with pid [14713]
process[playbag-5]: started with pid [14720]
[ INFO] [1487927180.139959185]: I0224 17:06:20.000000 14699 configuration_file_resolver.cc:40] Found '/home/yhzhao/catkin_ws/src/cartographer_ros/cartographer_ros/configuration_files/backpack_2d.lua' for 'backpack_2d.lua'.
[ INFO] [1487927180.140257572]: I0224 17:06:20.000000 14699 configuration_file_resolver.cc:40] Found '/home/yhzhao/catkin_ws/devel_isolated/cartographer/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1487927180.140309976]: I0224 17:06:20.000000 14699 configuration_file_resolver.cc:40] Found '/home/yhzhao/catkin_ws/devel_isolated/cartographer/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1487927180.140388145]: I0224 17:06:20.000000 14699 configuration_file_resolver.cc:40] Found '/home/yhzhao/catkin_ws/devel_isolated/cartographer/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1487927180.140429867]: I0224 17:06:20.000000 14699 configuration_file_resolver.cc:40] Found '/home/yhzhao/catkin_ws/devel_isolated/cartographer/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1487927180.140549279]: I0224 17:06:20.000000 14699 configuration_file_resolver.cc:40] Found '/home/yhzhao/catkin_ws/devel_isolated/cartographer/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1487927180.140596277]: I0224 17:06:20.000000 14699 configuration_file_resolver.cc:40] Found '/home/yhzhao/catkin_ws/devel_isolated/cartographer/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1487927180.140732901]: I0224 17:06:20.000000 14699 configuration_file_resolver.cc:40] Found '/home/yhzhao/catkin_ws/devel_isolated/cartographer/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1487927180.140775142]: I0224 17:06:20.000000 14699 configuration_file_resolver.cc:40] Found '/home/yhzhao/catkin_ws/devel_isolated/cartographer/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1487927180.173229746]: I0224 17:06:20.000000 14699 submaps.cc:176] Added submap 1
[ INFO] [1487927180.173314821]: I0224 17:06:20.000000 14699 map_builder_bridge.cc:37] Added trajectory with ID '0'.
F0224 17:06:20.730864 14699 laser.cc:46] Check failed: proto.range_min() >= 0.f (-2.35183 vs. 0) 
[FATAL] [1487927180.731154775, 1432647016.539545059]: F0224 17:06:20.000000 14699 laser.cc:46] Check failed: proto.range_min() >= 0.f (-2.35183 vs. 0) 
*** Check failure stack trace: ***
    @     0x7f26b3ca3daa  (unknown)
    @     0x7f26b3ca3ce4  (unknown)
    @     0x7f26b3ca36e6  (unknown)
    @     0x7f26b3ca6687  (unknown)
    @           0x5299b2  cartographer::sensor::ToPointCloudWithIntensities()
    @           0x529b00  cartographer::sensor::ToPointCloud()
    @           0x5019fe  cartographer_ros::SensorBridge::HandleMultiEchoLaserScanMessage()
    @           0x4eb89c  _ZN5boost6detail8function26void_function_obj_invoker1IZN16cartographer_ros12_GLOBAL__N_13RunEvEUlRKNS_10shared_ptrIKN11sensor_msgs19MultiEchoLaserScan_ISaIvEEEEEE0_vSD_E6invokeERNS1_15function_bufferESD_
    @           0x4f0b5b  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x4f73e8  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7f26b0b8f6b5  (unknown)
    @     0x7f26b0b49107  (unknown)
    @     0x7f26b0b49c33  (unknown)
    @     0x7f26b0b921e5  (unknown)
    @     0x7f26b0b79e0b  (unknown)
    @           0x4ec94b  cartographer_ros::(anonymous namespace)::Run()
    @           0x4e9d7d  main
    @     0x7f26af94cf45  (unknown)
    @           0x4eb427  (unknown)
    @              (nil)  (unknown)
[cartographer_node-3] process has died [pid 14699, exit code -6, cmd /home/yhzhao/catkin_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_node -configuration_directory /home/yhzhao/catkin_ws/src/cartographer_ros/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer_node __log:=/home/yhzhao/.ros/log/81843de6-fa70-11e6-8456-08626614ffb2/cartographer_node-3.log].
log file: /home/yhzhao/.ros/log/81843de6-fa70-11e6-8456-08626614ffb2/cartographer_node-3*.log
^C[playbag-5] killing on exit
[rviz-4] killing on exit
[robot_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

SirVer, 2017-02-24:
Please check that your bag file is not corrupt. SHASUM:

~~~
$ shasum ~/Downloads/cartographer_paper_deutsches_museum.bag                                                                                                                                         
2a021fadae9deb0643d73c8ca3acb332fcb3baa2
~~~

If you have the same checksum, something is fishy in your build. Which OS and which version is this? A pristine checkout I assume? Have you tried running in one of our docker container?

kintzhao, 2017-02-24:
del  [HECK_GE(proto.range_min(), 0.f);](https://github.com/googlecartographer/cartographer/blob/628b9da6d226ba9825f3bbf941fc8f4b98016a0f/cartographer/sensor/laser.cc#L46) 
    std::cout<<" proto.range_min() "<<proto.range_min()<<std::endl;
  //CHECK_GE(proto.range_min(), 0.f);
 this can run and build map, but the map is disorder， output  :
---------------------
[ INFO] [1487931149.671169969, 1432647106.745210471]: I0224 18:12:29.000000  6325 constraint_builder.cc:276] 818 computations resulted in 140 additional constraints.
[ INFO] [1487931149.671408867, 1432647106.745210471]: I0224 18:12:29.000000  6325 constraint_builder.cc:278] Score histogram:
Count: 2492  Min: 0.550073  Max: 0.900000  Mean: 0.721847
[0.550073, 0.585066)	                ####	Count: 520 (20.866774%)	Total: 520 (20.866774%)
[0.585066, 0.620058)	                 ###	Count: 350 (14.044944%)	Total: 870 (34.911716%)
[0.620058, 0.655051)	                  ##	Count: 203 (8.146068%)	Total: 1073 (43.057785%)
[0.655051, 0.690044)	                   #	Count: 142 (5.698235%)	Total: 1215 (48.756020%)
[0.690044, 0.725037)	                   #	Count: 115 (4.614767%)	Total: 1330 (53.370785%)
[0.725037, 0.760029)	                   #	Count: 108 (4.333869%)	Total: 1438 (57.704655%)
[0.760029, 0.795022)	                   #	Count: 146 (5.858748%)	Total: 1584 (63.563404%)
[0.795022, 0.830015)	                   #	Count: 135 (5.417336%)	Total: 1719 (68.980736%)
[0.830015, 0.865007)	                   #	Count: 126 (5.056180%)	Total: 1845 (74.036919%)
[0.865007, 0.900000]	               #####	Count: 647 (25.963081%)	Total: 2492 (100.000000%)
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
[ INFO] [1487931149.865636822, 1432647106.936535365]: I0224 18:12:29.000000  6325 sparse_pose_graph.cc:280] We caught up. Hooray!
 proto.range_min() -2.35183
[ INFO] [1487931149.874765688, 1432647106.946589414]: I0224 18:12:29.000000  6327 constraint_builder.cc:255] Scan index 1639 with 109 points on submap 16 differs by translation 0.08 rotation 0.016 with score 64.2% covariance trace 3.4462e-08.
[ INFO] [1487931149.883111227, 1432647106.956650493]: I0224 18:12:29.000000  6327 constraint_builder.cc:255] Scan index 1644 with 107 points on submap 16 differs by translation 0.04 rotation 0.022 with score 62.7% covariance trace 4.3090e-08.
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
[ INFO] [1487931150.002394206, 1432647107.070116870]: I0224 18:12:30.000000  6327 constraint_builder.cc:255] Scan index 1649 with 100 points on submap 16 differs by translation 0.13 rotation 0.031 with score 58.5% covariance trace 5.1605e-08.
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183
 proto.range_min() -2.35183

kintzhao, 2017-02-24:
@SirVer    yhzhao@X550JD:~/rosBagData$ shasum cartographer/cartographer_paper_deutsches_museum.bag 
c8ef93e8464c6d58a22fc737274a6504ee9f0838  cartographer/cartographer_paper_deutsches_museum.bag

rosBag is ok, I can run it well using cartoGrapher_ros [
commit 0c8d6afc33531761e8dd69760907d730a8440ed3
Author: Wolfgang Hess <whess@google.com>
Date:   Thu Oct 6 16:53:31 2016 +0200
]

ubuntu 14.04 64bits   i7-4710HQ   indigo

kintzhao, 2017-02-24:
How can I get the docker  url?

kintzhao, 2017-02-24:
@SirVer I download  the new rosbag , it can run ok.  but i still can not find the reason . does the  cartographer_paper_deutsches_museum.bag  record different ?

SirVer, 2017-02-24:
I do not understand your question. You redownloaded the ROS bag and now it works?

kintzhao, 2017-02-25:
yes, it works when redownloaded the ROS bag...   But the older ROS bag  can works,  ever describe as [#41](https://github.com/googlecartographer/cartographer_ros/issues/41)

SirVer, 2017-02-27:
I do not have an explanation why the bag on disk stopped working for you or got modified somehow... but as long as it works now.

wohe, 2017-02-27:
@kintzhao The old ROS bag was indeed broken. With googlecartographer/cartographer#114 and #157 I think, the `CHECK` you see failing was introduced. Before this change, the `range_min` in the bag was ignored by Cartographer. A fixed ROS bag having a positive value for `range_min` was uploaded at the same time.

closed 2017-02-27 by wohe

Lingbo2017, 2019-03-18:
Hello! I have met the same problem again, and my computer configuration is here:
**ubuntu 16.04, 64bit , kinetic.**
All of them went well in compiling process until I launch the 2D demo using this command:
**roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag**

The ERROR part came as follows:
[ INFO] [1552891680.239283186, 1432647016.574284759]: I0318 14:48:00.000000 14323 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '635682438164904058'.
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
[cartographer_node-3] process has died [pid 14323, exit code -6, cmd /home/lingbo/catkin_cartographer/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/lingbo/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer_node __log:=/home/lingbo/.ros/log/c41d31ce-4949-11e9-8afe-54271ed5aac5/cartographer_node-3.log].
log file: /home/lingbo/.ros/log/c41d31ce-4949-11e9-8afe-54271ed5aac5/cartographer_node-3*.log

Some has inferred that this is an alignment issue from running on [32-bit.](https://github.com/googlecartographer/cartographer_ros/issues/48)
**But I am pretty sure that both my cpu and ubuntu 16.04 OS are 64 bit.** 
So could anyone help me figure this out here? Thank you very much!