cartographer-project/cartographer_ros #271: compile point_cloud_viewer error

sunmk2006, 2017-03-03:
I want to build point cloud viewer as follows,
    1 Change into the client directory: cd client.
    2 Install npm. We strongly suggest using nvm.
    3 Install javascript dependencies: npm install.
    4 Build the client: npm run build.
There was some errors when I did the third step，
viki@viki:~/soft/point_cloud_viewer-master/client$ npm install
npm WARN package.json point_viewer@0.1.0 No repository field.
npm WARN package.json point_viewer@0.1.0 No README data
npm ERR! 404 Not Found
npm ERR! 404 
npm ERR! 404 'types/three' is not in the npm registry.
npm ERR! 404 You should bug the author to publish it
npm ERR! 404 It was specified as a dependency of 'point_viewer'
npm ERR! 404 
npm ERR! 404 Note that you can also install from a
npm ERR! 404 tarball, folder, or http url, or git url.

npm ERR! System Linux 4.2.0-27-generic
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/viki/soft/point_cloud_viewer-master/client
npm ERR! node -v v0.10.37
npm ERR! npm -v 1.4.28
npm ERR! code E404
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/viki/soft/point_cloud_viewer-master/client/npm-debug.log
npm ERR! not ok code 0
viki@viki:~/soft/point_cloud_viewer-master/client$

SirVer, 2017-03-03:
Thank you. The problem seems that npm has a different understanding of semver than I am used to - some packages went out of backwards compatibility and did not work together anymore.

closed 2017-03-03 by SirVer

sunmk2006, 2017-03-04:
build successfully, but：
viki@viki:/soft/point_cloud_viewer-master/target/release$ ls
build build_octree deps examples libpoint_viewer.rlib native viewer
viki@viki:/soft/point_cloud_viewer-master/target/release$ build_octree
build_octree: command not found
viki@viki:~/soft/point_cloud_viewer-master/target/release$

sunmk2006, 2017-03-04:
./build_octree generates a lot of files,but
./viewer /home/viki/source/data/cartographer_data/mode 
Listening on port 5433.
IT seems to do nothing,  No changes in memory and CPU ，What happened？