cartographer-project/cartographer_ros #273: How to config appropriately for VLP-16 laser in indoor case?

luoxiaoliaolan, 2017-03-03:
Hello,
I tried to use a Velodyne VLP-16 laser (I tried to modify cartographer_ros’s configuration file so that I can use a Velodyne VLP-16 laser to collect points) and a IMU to generate indoor 3D pointcloud, then get a precise indoor map with google cartographer. However I always got a set of clutter points and the process of showing cloudpoints on the screen is very slow.
The pointcloud are as below:
[image]
[image]

And I can’t get apparent trajectory map on these grids.
My related configuration files：
backpack_3d.urdf:
```
<robot name="cartographer_backpack_cnu_2d">
  <material name="orange">
    <color rgba="1.0 0.5 0.2 1" />
  </material>
  <material name="gray">
    <color rgba="0.2 0.2 0.2 1" />
  </material>
  <material name="green">
    <color rgba="0.2 0.4 0.2 1" />
  </material>

  <link name="imu_link">
    <visual>
      <origin xyz="0.0 0.0 0.0" />
      <geometry>
        <box size="0.06 0.04 0.02" />
      </geometry>
      <material name="orange" />
    </visual>
  </link>

  <link name="horizontal_laser_link">
    <visual>
      <origin xyz="0.0 0.0 0.0" />
      <geometry>
        <cylinder length="0.09" radius="0.03" />
      </geometry>
      <material name="gray" />
    </visual>
  </link>

  <link name="vertical_laser_link">
    <visual>
      <origin xyz="0.0 0.0 0.0" />
      <geometry>
        <cylinder length="0.09" radius="0.03" />
      </geometry>
      <material name="gray" />
    </visual>
  </link>

  <link name="base_link" />

  <joint name="imu_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="imu_link" />
    <origin xyz="-0.04 0.0315 0.016" rpy="0 0 0" />
  </joint>

  <joint name="horizontal_laser_joint" type="fixed">
    <parent link="base_link" />
    <child link="horizontal_laser_link" />
    <origin xyz="-0.0495 0.1695 0.081" rpy="1.57 0 0" />
  </joint>

  <joint name="vertical_laser_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="vertical_laser_link" />
    <origin rpy="1.57 -1.57 0" xyz="-0.14 0.241 0.037" />
  </joint>

</robot>
```

backpack_3d.lua:
```
include "map_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry_data = false,
  use_constant_odometry_variance = false,
  constant_odometry_translational_variance = 0.,
  constant_odometry_rotational_variance = 0.,
  use_horizontal_laser = false,
  use_horizontal_multi_echo_laser = false,
  horizontal_laser_min_range = 0.,
  horizontal_laser_max_range = 30.,
  horizontal_laser_missing_echo_ray_length = 5.,
  num_lasers_3d = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

TRAJECTORY_BUILDER_3D.scans_per_accumulation = 80

MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 7
MAP_BUILDER.sparse_pose_graph.optimization_problem.huber_scale = 5e2
MAP_BUILDER.sparse_pose_graph.optimize_every_n_scans = 320
MAP_BUILDER.sparse_pose_graph.constraint_builder.sampling_ratio = 0.03
MAP_BUILDER.sparse_pose_graph.optimization_problem.ceres_solver_options.max_num_iterations = 10
-- Reuse the coarser 3D voxel filter to speed up the computation of loop closure
-- constraints.
MAP_BUILDER.sparse_pose_graph.constraint_builder.adaptive_voxel_filter = TRAJECTORY_BUILDER_3D.high_resolution_adaptive_voxel_filter
MAP_BUILDER.sparse_pose_graph.constraint_builder.min_score = 0.62
MAP_BUILDER.sparse_pose_graph.constraint_builder.log_matches = true

return options
```

My dataset:
[Bag file](https://onedrive.live.com/?authkey=%21APZXmCCjV41MGyI&id=EEE7232656781728%216119&cid=EEE7232656781728)

SirVer, 2017-03-06:
> TRAJECTORY_BUILDER_3D.scans_per_accumulation = 80

Is this correct? If your laser gives you full revolutions (which is likely if you did not change it), you accumulate full 80 revolutions into one cloud before scan matching. [See this FAQ entry for more information](https://google-cartographer-ros.readthedocs.io/en/latest/faq.html#the-laser-data-in-the-3d-bags-is-much-higher-than-the-maximum-reported-20-hz-rotation-speed-that-the-vlp-16-can-do-why).

Also note: on the second picture you can see that the tiny opening angle of the VLP16 is not ideal for full 3D slamming (which you are trying to do): nothing constraints this point cloud is z-direction, i.e. it will likely drift in Z. You will likely require a second sensor to constrain you in the up down direction, but maybe (depending on the features of your environment) it works.

Let me know if you require further assistance.

luoxiaoliaolan, 2017-03-06:
My dataset
Bag file
(PS:The Laser frequency is 10Hz)

luoxiaoliaolan, 2017-03-06:
Today I record a set of data to test 3D slamming with a VLP-16 Laser , and the link is as below:
[My dataset](https://1drv.ms/u/s!AigXeFYmI-fur2mVur4cJEi4EnsN)
(Note:My laser frequency is 10Hz)
However I can't get ideal matching results, according to your advice.  Could you help me to test this data?
Thanks a lot.

SirVer, 2017-03-06:
Please provide a link to a `cartographer_ros` fork that contains your configuration files and launch files.

luoxiaoliaolan, 2017-03-06:
My fork link is: https://github.com/luoxiaoliaolan/cartographer_ros
The launch file is: [demo_backpack_my_3d.launch](https://github.com/luoxiaoliaolan/cartographer_ros/blob/master/cartographer_ros/launch/demo_backpack_my_3d.launch)

Thanks

SirVer, 2017-03-07:
I only had a quick look at your data, but there is obvious things wrong here that you need to fix:

- As mentioned above, your `scans_per_accumulation` should be 1 since you did not subdivide your point cloud. This is obvious by just trying this setting out.
- Your IMU data is wonky (see below): first, its covariances are all zero (which is wrong) and your linear acceleration does not contain gravity - in fact all values in your packets are basically 0 which is quite unphysical.

~~~
header: 
  seq: 16987
  stamp: 
    secs: 1488505542
    nsecs: 858946084
  frame_id: imu_link
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 6.30675e-15
  y: -1.02985e-14
  z: 1.41055e-14
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: -2.05425e-16
  y: 9.4725e-17
  z: 8.56125e-15
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
~~~

These are quite fundamental issues. Please fix them and read through past closed issues labeled 'tuning'. If you still have issues then, please open a new bug.

closed 2017-03-07 by SirVer