cartographer-project/cartographer_ros #274: 2D SLAM reconstruction

sunmk2006, 2017-03-04:
After the successful completion of the 2D map, get a track, when cartographer need to do reconstruction, whether or not to use the horizontal laser data,in 2D mapping?

damonkohler, 2017-03-06:
I'm sorry, I really don't understand the question. A horizontal-ish LIDAR fan is required for 2D mapping.

sunmk2006, 2017-03-06:
If I already have a perfect trajectory, and a vertically placed 2D laser, can use assets writer to get a 3D point cloud mode.Focus on reconstruction.

SirVer, 2017-03-07:
It sounds like you have everything? What are you missing? 

Example commandline to run the assets_writer over the paper data:
    
    roslaunch cartographer_ros assets_writer_backpack_2d.launch \
      bag_filenames:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag \
      trajectory_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag.pb

closed 2017-03-07 by SirVer