cartographer-project/cartographer_ros #275: Problem with Inclined Velodyne HDL32-E

mmazaherit, 2017-03-06:
Hello

I have a system which includes a Velodyne HDL-32E tightly attached to a IMU, but leaning 45.43 degrees backward (in the opposite direction of movement). I shared ( [this link](https://drive.google.com/file/d/0B1I7TzV8-JvyZzVxVmZybEVqT1k/view?usp=sharing) ) the settings and two datasets  collected indoor in two different poses: 

A) Natural pose in which Velodyne is inclined backward and IMU is horizontal
B) Tilted pose in which Velodyne is vertical and IMU is tilted

I was hopping to provide enough constraints to estimate 2D and 3D motion of this system, but so far I had only success with pose B (vertical velodyne) using cartographer in 2D, which nicely works and closes the loop. Now my questions are:

1-What settings do I miss that I can not run cartographer with inclined Velodyne? not even in 2D

2-In my opinion a Velodyne HDL-32E provides enough constraints in 3D, especially in an indoor environment rich in feature, but I couldn't get any (meaningful) result in 3D so far.

Any help from cartographer team is appreciated, as I'm out of idea

mmazat, 2017-04-24:
I need to update this data as I found a problem with IMU/Velodyne synchronization.  However, I expect 2D slam with Inclined Velodyne should work, as we can ignore the IMU

SirVer, 2017-04-24:
@mmazat Does this mean the data you linked to in the OP is no longer useful?

> However, I expect 2D slam with Inclined Velodyne should work, as we can ignore the IMU

I do not follow this reasoning. Could you expand a bit? 2D SLAM requires that the laser spin axis is perpendicular to the plane of movement - otherwise there is no overlap between the individual laser revolutions.

mmazat, 2017-04-24:
Thanks a lot for the responses.

Right, our Velodyne is highly inclined, I should not expect result in 2D.

Yes, I found out IMU data in this data has problem, I have better setup now to acquire a better dataset.

SirVer, 2017-04-24:
I will hold off on looking into this issue then.

SirVer, 2017-04-26:
Closing this, since it is not actionable for us right now. @mmazat If you have more questions or tuning problems, please open a new issue and refer this one.

closed 2017-04-26 by SirVer