cartographer-project/cartographer_ros #276: Lightweight Configuration for Raspberry Pi

ruffsl, 2017-03-07:
I'm running cartographer_ros (or more specifically cartographer_turtlebot), using a modified version of 
[turtlebot_urg_lidar_2d.lua](https://github.com/ruffsl/fetchbot_pi/blob/08dc27382df84abacc226cf2b41dc5d44843bd27/fetchbot/configuration_files/turtlebot_urg_lidar_2d.lua), on a Raspberry Pi 3B. This little guy only has 4 cores, and 1GB of RAM, so getting cartographer to run more that a minute in a small hallway before map TF rates drop below the navigation stack's max of 10sec delay tolerance is impossible with my current config. 

Are there any reasonable compromises you'd suggest for configuration files targeting embedded systems to achieve reasonable and long lived publication rates? The space I'm working with is relatively small. I know I could should build a map once offline, then service a static map up along with AMCL for localization, but I wa wondering if there'd still be a way to keep cartographer running in a lightweight setup for continuous SLAM.

So far I've enlarged the `laser_voxel_filter_size` and `optimize_every_n_scans`, as well as lowered the `submaps.resolution` and `motion_filter.max_time_seconds`.

SirVer, 2017-03-07:
> Are there any reasonable compromises you'd suggest for configuration files targeting embedded systems to achieve reasonable and long lived publication rates?

Not really. Right now, Cartographer is always SLAMing, so memory footprint will monotonically increase as Submaps are added. We have plans to support localization only with Cartographer eventually, but there is no timeline yet (help is appreciated :)).

> laser_voxel_filter_size, optimize_every_n_scans

This buys CPU, so this is a useful change. 

> submaps.resolution

This is probably the most impactful change. it buys memory and CPU. 

Some more assorted thoughts:

- ARM is not a supported architecture from us. We also do not support 32bit. Congrats for even make this work. Are there patches you can contribute back?
- In past issues, complaints about performance of Cartographer was very often tied to
    - Ceres not using libsparse
    - Eigen, ceres, libsparse or Cartographer not being build in opt mode (each of these can make an order of magnitude difference).

  Cartographer is complaining if you do not build it in opt mode these days, but the others can still be an issue.
- My expectation for the Raspberrypi 3 is that it should work better/longer than what you describe - given hardware and RAM, so I would start investigating if everything is properly optimized. Then again, as said we never tested on ARM/32bit systems so my experience with 64bit/x86_64 maybe does not translate well to your environment - i.e. maybe that is the best you can expect.

Minipada, 2017-03-20:
I have managed by compiling manually libatlas, libblas, libmetis (in 32 bits mode on the rpi) to run Cartographer on the rpi at 1 Hz, 2 days ago. the map was successfully created.
Note that those compilation took more than 3 days to complete and it crashed at 2 Hz. 

I agree you need to use libsparse. I would advise you to use a custom OS. Mine is pretty optimized and I think it will not run at all on a heavy raspbian

I am now investigating to cross compile all those optimization libs, but I'm a bit lost to be honest. Any help is welcomed

SirVer, 2017-03-20:
Congratulations! That is quite the achievement.

I am closing this bug as we will not officially support the RaspberryPi going forward. Please keep us posted with your experiments though!

closed 2017-03-20 by SirVer

ruffsl, 2017-03-20:
@Minipada , I "cross compiled" this project by compiling everything using a docker container, with the image being an armhf release of ubuntu with ros installed, and qemu enabled on the host and with the static qemu target added into the container. Didn't take nearly as long as 3 days to compile, but it did take an hour or so. Once I get some more time next week I'll try and write more on it, but for now check out this [issue](https://github.com/osrf/multiarch-docker-image-generation/issues/6#issuecomment-282943316), and my old [Dockerfile](https://github.com/ruffsl/fetchbot_pi/blob/b526748d736e30e8176b6834540cc64b65806f4d/docker/cartographer_ros/Dockerfile). I've pushed the image here: [ruffsl/ros_armhf:kinetic_cartographer_ros](https://hub.docker.com/r/ruffsl/ros_armhf/tags/), but it may grow stale or I may move it.

Minipada, 2017-03-21:
@ruffsl Those libraries don't need to be compiled in 3 days, for sure. From what I understood, it needs to be compiled on the rpi to get the best optimizations as they mention it [here](http://www.vesperix.com/arm/atlas-arm/faq/index.html). I did using Yocto, more complex than docker indeed.

ruffsl, 2017-03-21:
@Minipada , from your link:

> #### Q3. Can I cross compile ATLAS?
> A3. No, and it wouldn't be a good idea even if you could. ATLAS chooses the fastest routines for inclusion in the BLAS library on the basis of tests it makes as the library builds. If you're not building the library on a machine similar to the one you want to use the BLAS on, then all that tuning is meaningless.

Hmm, nice find, didn't know about this project. So how are you rolling out the cartographer library (and friends) using [ATLAS](http://math-atlas.sourceforge.net/faq.html) (Automatically Tuned Linear Algebra Soft) though? I just briefly read about the project, and wasn't sure how you've included it into the compilation.

I'm also new to the Yocto project, but would it be possible to retain a learned build profile from the raspberry pi target via a Board Support Package, and use that during cross compilation?

Minipada, 2017-03-21:
@ruffsl Thanks. I have compiled on a custom generated OS metis, lapack, blas and atlas and added the generated .so & .a in the Yocto workflow. Then I have written recipes for cartographer, cartographer-ros, cartographer-ros-msgs and ceres-solver, they are cross compiled in the standard Yocto workflow and use the generated libraries.
It is not the standard way because it means I can only use them for this particular architecture. If for any reason I'd change to a new board or a 64 bits OS, I'd have to do it again.

I didn't find a conservative way, even tho it seems there are starting points, mentioned [here](https://www.mail-archive.com/search?l=yocto@yoctoproject.org&q=subject:%22%5C%5Byocto%5C%5D+Package+depending+on+ATLAS%2C+BLAS%2C+libmetis%2C+liblapack%22&o=newest&f=1) and [here](https://github.com/MacPython/numpy-atlas-binaries/blob/master/README.md).