cartographer-project/cartographer_ros #277: Generate and view 2G ply point cloud data, how long it takes, how much memory

sunmk2006, 2017-03-13:
Generate 2G size point cloud data, ply file，how long it takes, how much memory，DO you have impression drawing of a 3D model?

SirVer, 2017-03-13:
I do not understand the question. What is 2G? 

Generating a PLY file is cheap in memory and processing though, since the data only streams through memory into the file. 

Please let me know if you have additional questions.

closed 2017-03-13 by SirVer

sunmk2006, 2017-03-13:
2G is 2GB(gigabyte)  .ply file size,

Generating ply files is very slow and expensive to my computer(cpu: i5 memory: 16GIB)

SirVer, 2017-03-13:
you probably also generate X-Rays in the same go. They are way more expensive. Comment out all `write_xray_image` generation steps in your [assets writer configuration file](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/configuration_files/assets_writer_backpack_2d.lua#L37).

sunmk2006, 2017-03-13:
I see. I'll give it a try