cartographer-project/cartographer_ros #279: messy map

gavinwtaylor, 2017-03-15:
I'm a professor, and I'm trying to get a map built of our hallway, hoping to use it for some cool projects later in the course.  I'm using turtlebots with depth sensors, and unfortunately, I'm not getting a very good map.  I was hoping somebody with some insight into configuration settings would be willing to help me out.  The run with the default settings was pretty bad, but given that it was a trajectory with a lot of turns, I turned up the motion filter's max_angle_radians to .8, and got the attached, which, believe it or not, is actually much better.  [All configuration files can be found here](https://github.com/gavinwtaylor/cartographyFiles).  [The bag file can be downloaded here](https://www.usna.edu/Users/cs/taylor/turny.bag) (~1G).

Any help in saving my class would be greatly appreciated!

SirVer, 2017-03-20:
Your Git directory does not contain any history and is not a fork of the Cartographer repositories. So it is unclear what changes you actually did to your configuration. Consequently, it is unclear to me which launch file and configuration you actually used for running your tests.

I'd rather not reverse engineer your URDF and your frames, so I will not further investigate for now. Please provide launch commands, urdf, configuration files and ideally a fork for cartographer and cartographer_ros so that I can see what you tried already.

SirVer, 2017-04-24:
Closing as stale.

closed 2017-04-24 by SirVer