cartographer-project/cartographer_ros #280: Do you have some paper about 3D cartographer?

sunmk2006, 2017-03-20:
Thank you for filing an issue!

It would help us tremendously if you provid the following so that we can easily
reproduce your issue:

1. A link to a Git repository containing a branch of `cartographer_ros`
   containing all the configuration, launch, and URDF files required to
   reproduce your issue.
2. A link to a bag file we can use to reproduce your issue.

We will likely be unable to help you without both.

Please remove this boilerplate text before submitting your issue.

damonkohler, 2017-03-20:
We have not yet written a paper about our 3D implementation.

closed 2017-03-20 by damonkohler

msy22, 2017-04-18:
Does that mean you are planning to write a paper on the 3D implementation? If so, what would be the ETA? If not, is there any available material on the 3D implementation? (other than source code).

Thanks in advance for any help or advice.

SirVer, 2017-04-18:
We plan to write a paper, but have no ETA and no cycles to spent on it currently. 

There is no other documentation besides the source code, but the 3D implementation is a straightforward extension of the 2D approach, so the source code is probably the most educational.

msy22, 2017-04-18:
Fantastic, thanks for the help!