cartographer-project/cartographer_ros #281: What is the difference between the online and offline cartographer?

sunmk2006, 2017-03-22:
What is the difference between the online and offline cartographer  in addition to data acquisition? The key is the algorithm flow and processing speed！

ojura, 2017-03-22:
As you mentioned, the main difference is in data acquisition - the online node subscribes to laser, imu, odometry, etc. data and runs in real time. 

Because the offline node reads all of this data directly from the bag, it sets its own pace and goes through the bag as quickly as it can. If you attempted to do this with the online node and e.g. you played the bag faster using the `-r` parameter of `rosbag play`, in case you played it too fast, you would risk missing messages in Cartographer. You would have to find the highest bag playing rate with which Cartographer would not miss messages. However, it does not really make sense to do all of this, and you should use the offline node instead, which will always process the bag at maximum speed.

sunmk2006, 2017-03-22:
Thank you for your quick reply ， I think I understand what you mean！
 But the perfomance is different between online and offline data according to the paper about spa, Whether also have an impact on the accuracy of the map in cartographer?

SirVer, 2017-03-22:
@ojura nailed it! That is exactly right.

> But the perfomance is different between online and offline data according to the paper about spa, Whether also have an impact on the accuracy of the map in cartographer?

I am not sure which paper you are referring to. Our paper does not distinguish between online and offline since it has no impact on quality.

closed 2017-03-22 by SirVer

sunmk2006, 2017-03-22:
K. Konolige, G. Grisetti, R. K ̈ummerle, W. Burgard, B. Limketkai,
and R. Vincent, “Sparse pose adjustment for 2D mapping,” in IROS,
Taipei, Taiwan, 10/2010 2010

VI. EXPERIMENTS

SirVer, 2017-03-22:
We based our implementation on this paper, but there is no code reuse. I.e. their experiments are completely unrelated to our implementation.

mohamadfarhat, 2017-04-03:
Hey, I have the same question. The Offline node read the data only faster? That mean the map quality at the end are the same between online and offline cartographer. How about the loop closure detection? they use the same algorithm or it is any difference? thx

SirVer, 2017-04-03:
Yes, the offline just processes data as quickly as possible by reading it directly from the bag. This allows for faster experiments, but is not as flexible as online.

tiandaji, 2018-11-15:
Hello! @ojura @SirVer @sunmk2006 @mohamadfarhat 
I am new on cartographer, and I would like to change the playing rate of the bag.
According to the manual, I managed to play the bag via the command:
ex. roslaunch cartographer_ros offline_backpack_2d.launch **bag_filenames**:=${HOME}/Downloads/b2-2016-04-05-14-44-52.bag
In fact, I want to follow @ojura 's suggestion, using ex. "rosbag play file_path_and_name.bag -r 1.2" to run the SLAM process 0.2 times faster than real time. But when I leave the **bag_filenames** empty in the roslaunch command, there will be a mistake, reporting no bag file used (which is obvious). 
Therefore, I suppose that I should  somehow turn off the **bag_filenames** part in the 1st example, and launch another terminal session to run "rosbag play" in the rate I desire. 
Am I right? And is there an easy way to achieve this?
Thanks a lot in advance.

ojura, 2018-11-15:
@tiandaji, the __offline__ node processes the bag at the maximum speed your CPU can process. You can not set the rate, it is determined by your CPU's performance. If you want to control the speed, you can use the __online__ node + `rosbag play -r 1.2`.

tiandaji, 2018-11-15:
@ojura Thx so much, I got it.