cartographer-project/cartographer_ros #282: motion state

sunmk2006, 2017-03-22:
when i use cartographer to do a 2D/3D slam, where is the boundary of the motion state of the device , Can it be used in vehicle systems?

SirVer, 2017-03-22:
I do not understand the question. Are you asking what devices Cartographer can work on? We used Cartographer on cars, backpacks, bipedal, quatropedal and wheeled robots as well as quatrocopters.

closed 2017-03-22 by SirVer

sunmk2006, 2017-03-22:
Can you tell me the speed of the vehicle？

SirVer, 2017-03-22:
Highway driving and inner city driving.

sunmk2006, 2017-03-22:
can cartographer  be  used for outdoor trajectory calculation？

SirVer, 2017-03-22:
Yes.

sunmk2006, 2017-03-22:
I am very worry about the motion distortion,do you have some answer for this question?

SirVer, 2017-03-22:
Rightfully so. You have to take that into account. We did that for our example bags, see our [FAQ entry](https://google-cartographer-ros.readthedocs.io/en/latest/faq.html#the-laser-data-in-the-3d-bags-is-much-higher-than-the-maximum-reported-20-hz-rotation-speed-that-the-vlp-16-can-do-why). Cartographer SLAM does proper unwarping to fix for the motion distortion.