cartographer-project/cartographer_ros #285: Cartographer is not using the odometry input

NIN11, 2017-03-22:
Hello, I've been using google cartographer for a couple of months, and it's been great, the only issue I have is with very regular hallways (classic slam issue), so I've been trying to add odometry to the node to achieve better results, but I've not been succesful so far, maybe you could help me tune it correctly. My sensor is a URG-04LX_UG01 ( http://www.hokuyo-aut.jp/02sensor/07scanner/download/pdf/URG-04LX_UG01_spec_en.pdf ), im not using an IMU.

This is the best result I have had in the trouble area (by the way, everywhere else it works great), the red line is the odometry published by the rosaria node on the topic odom:
[image]

Here is the repo, with the launch files, configuration files and URDF:
https://github.com/NIN11/cartographer_ros

My two bag files, one for each launch file I added:
https://drive.google.com/file/d/0B8ONeY2eleY3dl92b3MyM3lQa0E/view?usp=sharing
https://drive.google.com/file/d/0B8ONeY2eleY3Y3MxUTc1RUlRTGM/view?usp=sharing

Any help would be greatly appreciated, and if you could tell me how to increase the rate at which the /map topic is published that would be great.

SirVer, 2017-03-23:
rosbag info tells me that your bags only contain `/odom`, `/slam_out`, and `/tf` topics. There is no laser scan in there.

SirVer, 2017-03-23:
This was quite challenging and showed week points in Cartographer. As usual, my changes are [in a branch in my fork](https://github.com/SirVer/cartographer_ros/commits/user/NIN11). The final map I got is below. I only looked at one of the bags you provided. The commandline I used is`roslaunch cartographer_ros replay_sin_mapa.launch bag_filename:=$HOME/Downloads/filtered_sin_mapa.bag`.

- Your sensor data is not great: Looking at your bag using rqt_bag shows sometimes 0.5 seconds between laser scans and odometry is similarly unreliable. Are you using a USB hub in between your sensors and your logging computer or something else that could introduce delays?
- You seem to be using `calibrate_time` for the URG node. I have had bad experiences with this setting - laser scan (non monotonous laser time stamps and jitter). You are probably better off turning it off.
- Your odometry has a bias: If you look at your red line in your picture you can see that one of your wheels is either bigger or turning faster than reported which makes your trajectory bend inwards. Generally it is best for SLAM with odometry to drive your path in each direction to make this bias observable.
- Loop closing is challenging: Cartographer finds lots of wrong constraints with really high score in the similar looking corridors. Also, your bag contains very little overlap (only the last few seconds are overlapping with the first few seconds), so Cartographer finds little correct loop closing constraints. So loop closure is really fickly on this dataset and easily messes everything up. Cartographer should probably not accept constraints that can be arbitrarily shifted along a line without the score changing, but for now it does. You can help it by providing more overlapping data - for example by driving the same trajectory backwards again. This should also help with odometry biases.
- Loop closing is further made more difficult by two rooms that are very similar being right next to each other (see red circles). It would have helped to explore the right most room more to get more data that discriminates it. This is also a week point of Cartographer: very similar looking rooms close to each other are matched and are hard to identify as outliers. The more overlapping data you collect, the more evidence you provide the algorithm to give up such a bad constraint though.

[image]

closed 2017-03-23 by SirVer

NIN11, 2017-03-26:
Hello again, and thank you. I'm seriously impressed, this is an amazing improvement, I wasn't expecting such a huge change in the output. 

I'll be using Google Cartographer as one of the nodes of my undergraduate thesis project, and with these changes I'll be able to do a lot more tests in this type of environments. I'll be sure to thank you in my thesis, you're a life saver. 

Thank you so much for your help. I really appreciate it.

caoxiongkun, 2017-08-01:
Hello SirVer! I have met the same problem,when I compile your fork,there is an error:

CMake Error at cartographer_ros/cartographer_ros_msgs/CMakeLists.txt:26 (google_initialize_cartographer_project):
  Unknown CMake command "google_initialize_cartographer_project"

Can you tell me how to solve it，Please？

ojura, 2017-08-01:
@caoxiongkun that forks is 4 old, that is ancient in Cartographer's time :-) You should rebase these commits from SirVer's fork (it's just a couple of config file tweaks) on top of the current master.