cartographer-project/cartographer_ros #287: It seems not have a loop closure, when the robot come back to the starting point

YuxiHuang001, 2017-03-27:
Hi, I’m trying to use cartographer_ros to build a map, the size of scene is about 200*100m. But the result seems not good,  especially when robot come back to the starting point, I didn't see any loop closure sign.
So I want to know if I set  some parameters  not appropriately. The package I used is 
https://github.com/YuxiHuang001/cartographer_ros
I run the demo_rover_2d.launch in the cartographer_ros package.
The result of mapping is in this package, the map_with_odom.png is the screenshot of the mapping, the odom trajectory is represented by red. The cartographer.pgm is the map saved by map_server it seems not good.
The bag file is
https://drive.google.com/file/d/0B55R0NT_50QKcmlCNFpFTEVRZEU/view?usp=sharing

The configuration of my computer is Intel® Core™ i7-5500U CPU @ 2.40GHz × 4 , 11.4GiB Memory, ubuntu14.04 ros indigo version.

Thx a lot~

SirVer, 2017-04-26:
Your fork does not contain any common history with `cartographer_ros`, i.e. it is not a fork at all. This makes it hard to see what you actually changed. Looking at the files it was based on a very old version of `cartographer_ros` - so you should probably pull at some point. I started my investigations with the current HEAD of `cartographer_ros` and copied your launch files and configuration from the repository you posted. They required minimal adaption to work on the current `HEAD`, but no tuning took place in this initial commit.

Your setup is quite strongly violating the 2D assumptions: neither is your floor flat (there are ramp-like ups and downs), nor are your walls perpendicular (some stuff at the side of the road look like shrubbery). Also your platform is not stable, i.e. the laser is not held perpendicular to the plane of movement at all times - and you do not provide IMU data for the system to compensate for this. These facts are not show stoppers for using 2D SLAM, but you should adjust your quality expectations accordingly. These violations are all nicely visualized if you set `TRAJECTORY_BUILDER_2D.submaps.range_data_insererter.insert_free_space` to `false` (see image below).

That said, I took your settings and only [tuned them for processing speed](https://github.com/SirVer/cartographer_ros/commit/5e3261f36036ebd6590ffcb5f7ccc7893e0c65f8) in the las commit [in this branch](https://github.com/SirVer/cartographer_ros/tree/user/YuxiHuang001) and the map that came out looks quite good to me (see below). The remaining slight duplication of walls is due to the violation of the 2D assumptions mentioned earlier. Your [result in the repo you posted](https://github.com/YuxiHuang001/cartographer_ros/blob/master/cartographer.pgm) looks worse. My guess is that Cartographer's quality improved since you downloaded it OR that your running in debug mode and your system is not realtime, ergo your map is without full loop closing having finisehd.

Map:
[image]

Map with `insert_free_space = false`. Note that the SLAM result is not exactly the same since this setting affects it also. Every smearing you see here is an indication of the violation of the 2D assumption - often ramps or non-horizontal walls.
[image]

closed 2017-04-26 by SirVer

YuxiHuang001, 2017-04-27:
OK,really appreciated