cartographer-project/cartographer_ros #288: pose_tracker.cc compile error

seanXYZ, 2017-03-28:
==> Processing plain cmake package: 'cartographer'
==> Creating build directory: 'build_isolated/cartographer/install'
==> Building with env: '/home/sean/catkin_carto/install_isolated/env.sh'
==> cmake /home/sean/catkin_carto/src/cartographer -DCMAKE_INSTALL_PREFIX=/home/sean/catkin_carto/install_isolated -G Ninja in '/home/sean/catkin_carto/build_isolated/cartographer/install'
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Build type: Release
Files /home/sean/catkin_carto/build_isolated/cartographer/install/AllFiles.cmake and - differ
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   iostreams
-- Found required Ceres dependency: Eigen version 3.2.0 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Performing Test GFLAGS_IN_GOOGLE_NAMESPACE
-- Performing Test GFLAGS_IN_GOOGLE_NAMESPACE - Success
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.12.0 installed in: /home/sean/catkin_carto/install_isolated with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, SchurSpecializations, OpenMP]
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0") 
-- Found Lua: /usr/lib/x86_64-linux-gnu/liblua5.2.so;/usr/lib/x86_64-linux-gnu/libm.so (found version "5.2.3") 
-- Found Threads: TRUE  
-- Found PROTOBUF: /usr/local/lib/libprotobuf.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for one of the modules 'cairo>=1.12.16'
-- Found Sphinx: /usr/bin/sphinx-build  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sean/catkin_carto/build_isolated/cartographer/install
==> ninja -j8 -l8 in '/home/sean/catkin_carto/build_isolated/cartographer/install'
[42/229] Building documentation.
FAILED: cd /home/sean/catkin_carto/build_isolated/cartographer/install/docs && /usr/bin/sphinx-build -b html /home/sean/catkin_carto/src/cartographer/docs/source /home/sean/catkin_carto/build_isolated/cartographer/install/docs/html
Traceback (most recent call last):
  File "/usr/bin/sphinx-build", line 18, in <module>
    sys.exit(main(sys.argv))
  File "/usr/lib/python2.7/dist-packages/sphinx/__init__.py", line 62, in main
    from sphinx import cmdline
  File "/usr/lib/python2.7/dist-packages/sphinx/cmdline.py", line 22, in <module>
    from sphinx.application import Sphinx
  File "/usr/lib/python2.7/dist-packages/sphinx/application.py", line 22, in <module>
    from docutils.parsers.rst import convert_directive_function, \
  File "/usr/lib/python2.7/dist-packages/docutils/parsers/rst/__init__.py", line 75, in <module>
    from docutils.parsers.rst import states
  File "/usr/lib/python2.7/dist-packages/docutils/parsers/rst/states.py", line 116, in <module>
    from docutils.parsers.rst import directives, languages, tableparser, roles
  File "/usr/lib/python2.7/dist-packages/docutils/parsers/rst/roles.py", line 78, in <module>
    from docutils.utils.code_analyzer import Lexer, LexerError
  File "/usr/lib/python2.7/dist-packages/docutils/utils/code_analyzer.py", line 13, in <module>
    from pygments.lexers import get_lexer_by_name
  File "/usr/lib/python2.7/dist-packages/pygments/lexers/__init__.py", line 18, in <module>
    from pygments.plugin import find_plugin_lexers
  File "/usr/lib/python2.7/dist-packages/pygments/plugin.py", line 39, in <module>
    import pkg_resources
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 72, in <module>
    import packaging.requirements
  File "/usr/local/lib/python2.7/dist-packages/packaging/requirements.py", line 59, in <module>
    MARKER_EXPR = originalTextFor(MARKER_EXPR())("marker")
TypeError: __call__() takes exactly 2 arguments (1 given)
[42/229] Building CXX object CMakeFiles/cartographer.dir/cartographer/kalman_filter/pose_tracker.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['/home/sean/catkin_carto/install_isolated/env.sh', 'ninja', '-j8', '-l8']' returned non-zero exit status 1

wohe, 2017-03-28:
According to the stack trace it is not the compilation of `pose_tracker.cc` that failed, but the execution of `/usr/bin/sphinx-build`. What happens if you just run `sphinx-build` on the command line?

seanXYZ, 2017-03-29:
sean@sean:~/catkin_carto$ /usr/bin/sphinx-build 
Traceback (most recent call last):
  File "/usr/bin/sphinx-build", line 18, in <module>
    sys.exit(main(sys.argv))
  File "/usr/lib/python2.7/dist-packages/sphinx/__init__.py", line 62, in main
    from sphinx import cmdline
  File "/usr/lib/python2.7/dist-packages/sphinx/cmdline.py", line 22, in <module>
    from sphinx.application import Sphinx
  File "/usr/lib/python2.7/dist-packages/sphinx/application.py", line 22, in <module>
    from docutils.parsers.rst import convert_directive_function, \
  File "/usr/lib/python2.7/dist-packages/docutils/parsers/rst/__init__.py", line 75, in <module>
    from docutils.parsers.rst import states
  File "/usr/lib/python2.7/dist-packages/docutils/parsers/rst/states.py", line 116, in <module>
    from docutils.parsers.rst import directives, languages, tableparser, roles
  File "/usr/lib/python2.7/dist-packages/docutils/parsers/rst/roles.py", line 78, in <module>
    from docutils.utils.code_analyzer import Lexer, LexerError
  File "/usr/lib/python2.7/dist-packages/docutils/utils/code_analyzer.py", line 13, in <module>
    from pygments.lexers import get_lexer_by_name
  File "/usr/lib/python2.7/dist-packages/pygments/lexers/__init__.py", line 18, in <module>
    from pygments.plugin import find_plugin_lexers
  File "/usr/lib/python2.7/dist-packages/pygments/plugin.py", line 39, in <module>
    import pkg_resources
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 72, in <module>
    import packaging.requirements
  File "/usr/local/lib/python2.7/dist-packages/packaging/requirements.py", line 59, in <module>
    MARKER_EXPR = originalTextFor(MARKER_EXPR())("marker")
TypeError: __call__() takes exactly 2 arguments (1 given)

seanXYZ, 2017-03-29:
sean@sean:/usr/bin$ ls sphinx-build -l 
lrwxrwxrwx 1 root root 44 11月  4 08:58 sphinx-build -> ../share/sphinx/scripts/python2/sphinx-build
sean@sean:/usr/bin$ 
sean@sean:/usr/share/sphinx/scripts/python2$ ls
sphinx-apidoc  sphinx-autogen  **sphinx-build**  sphinx-quickstart
sean@sean:/usr/share/sphinx/scripts/python2$

sean0822, 2017-03-29:
I reinstall sphinx-build, problem is removed! please close the topic @wohe

closed 2017-03-29 by wohe