cartographer-project/cartographer_ros #290: FixZ localparametrization

mmazat, 2017-03-28:
Hi there, 

I realized you have  FixZ local Parametrization in the code, but can't be activated in trajectory_build_3d.lua .  I wish we had this feature along with only_optimize_yaw for constraining 3D data to estimate 2D motion.

SirVer, 2017-04-24:
We played around with constraining Z in some experiments but didn't like it since it is a rather unprincipled approach and we also did not achieved satisfying results. Have you tried hardcoding this into the code and see if it works for you? We would be very interested in other experiences with this.

I agree though that we should either remove it from the code or somehow make it accessible in the configuration files. @wohe, thoughts?

mmazat, 2017-04-24:
We are experiencing a slight height difference between both end of a long loop, which might be the reason that the loop is not closed. I will try hard coding FixZ and hopefully this height difference goes away.

However, I have tried FixZ and only_optimize_yaw for some experiments before, and didn't see any change/improvement, but I'm not conclusive yet, and will share if I get any good result.

SirVer, 2017-05-15:
@mmazat I am closing this issue, since there is nothing actionable here for us. If you have anything new to share, please still update us here.

closed 2017-07-20 by damonkohler