cartographer-project/cartographer_ros #295: add IMU error

sean0822, 2017-03-30:
before I add the IMU， the result is quite good. 
After adding IMU, I found the laser is  sloping, about 15 degree. When I rotate my equipment, the laser also move, but not in the way i want .  I have added IMU for hector_slam, everything works fine. But this time, it does not work! BELOW is my Problem:

I checked the IMU data in another rviz window, it seems OK, but I don't known what 's the problem

[image]

`<launch>

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename rplidar_imu.lua"
      output="screen">
    <remap from="~scan" to="horizontal_laser_2d" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" required="true" args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />

  <node name="tf_base_link2horizontal_laser_link" type="static_transform_publisher" pkg="tf" args="0 0 0 3.14159 0 0  /base_link /horizontal_laser_link  40" />

  <node name="tf_base_link2imu_link" type="static_transform_publisher" pkg="tf" args="0 0 0 0 0 0 /base_link /imu_link  40" />

</launch>`

sean0822, 2017-03-30:
<launch>

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename rplidar_imu.lua"
      output="screen">
    <remap from="~scan" to="horizontal_laser_2d" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" required="true" args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />

  <node name="tf_base_link2horizontal_laser_link" type="static_transform_publisher" pkg="tf" args="0 0 0 3.14159 0 0  /base_link /horizontal_laser_link  40" />

  <node name="tf_base_link2imu_link" type="static_transform_publisher" pkg="tf" args="0 0 0 0 0 0 /base_link /imu_link  40" />

</launch>

sean0822, 2017-03-30:
here is my TF about the imu_link

node name="tf_base_link2imu_link" type="static_transform_publisher" pkg="tf" args="0 0 0 0 0 0 /base_link /imu_link  40"

SirVer, 2017-04-24:
Please provide a fork of `cartographer_ros` with your configuration files added and an example bag that shows the problem.

damonkohler, 2017-05-12:
Sounds like this is a problem with an incorrect transform between the IMU and the LIDAR. Closing due to inactivity.

closed 2017-05-12 by damonkohler