cartographer-project/cartographer_ros #297: Recommended LiDAR for 3D mapping with Cartographer

msy22, 2017-04-07:
I'm considering using Cartographer for my research into 3D mapping. I see that a lot of other people have asked about using their LiDAR unit for mapping. The Cartographer team has cautioned against some devices (e.g. [Velodyne VLP-16 because of its small vertical FOV](https://github.com/googlecartographer/cartographer/issues/213)). So I wanted to ask:

What LiDAR device (or setup of devices) does the Cartographer team recommend for 3D mapping with Cartographer?  

Thanks in advance for any suggestions or advice.

SirVer, 2017-04-07:
We have limited experience, since we only tested a few, so I would be hesitant to give buying advice here. That said, I feel there is no perfect SLAM laser available on the free market right now. 

We used all Velodyne devices for mapping before and they work well in pairs (i.e. one horizontal and one vertical) to work around the small opening angle. The more-beams ones have a better opening angle so one could be enough.

Hokuyo lasers work really well for 2D mapping and they have a 3D offering now, but we never tested them ourselves.

Generally you want a point could that constraints the SLAM problem. A laser with a big opening angle will do this nicely, while one with a small one will only constraint you only in the plane of movement. But you can combine multiple sensors (i.e. add a depth camera) to constrain the dimension that your laser is not doing for you. This of course comes with a different set of challenges: extrinsic calibration and consistent timing between sensors.

closed 2017-04-07 by SirVer

msy22, 2017-04-09:
Thanks for the advice!