cartographer-project/cartographer_ros #298: add IMU issue

cyy12345, 2017-04-08:
When I change the TRAJECTORY_BUILDER_2D.use_imu_data = false into true,I just find my cartographer can't run well. There is no map formed. I don't know what to do if I want to add the IMU data into cartographer node

SirVer, 2017-04-10:
Have you investigated URDF tree issues? Have you made sure the frame name's are correct? Is the topic properly remapped? What is `roswtf` reporting?

cyy12345, 2017-04-10:
Sorry, I can't give you the screenshot of my rviz, because I don't have my laser now. 
I think the main problem is that it shows "frame map does not exist".
And my launch file has   <remap from="scan" to="scan" />  I don't know is it correct

cyy12345, 2017-04-10:
<launch>  
  <param name="/use_sim_time" value="true" />

  <node name="robot_state_publisher" pkg="robot_state_publisher"
    type="robot_state_publisher" />

  <node name="cartographer_node" pkg="cartographer_ros"  
      type="cartographer_node" args="  
          -configuration_directory $(find cartographer_ros)/configuration_files  
          -configuration_basename revo_lds.lua"  
      output="screen">  
    <remap from="scan" to="scan" />  
  </node>

  <node name="rviz" pkg="rviz" type="rviz" required="true"  
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" /> 

 
</launch>

SirVer, 2017-04-10:
Sounds like a ROS wiring issue to me, not per se Cartographer related. Did you have success asking on answers.ros.org with this?

SirVer, 2017-04-10:
Closing as this is likely a ROS wiring issue, not a Cartographer issue. If you disagree, I will reopen.

closed 2017-04-10 by SirVer