cartographer-project/cartographer_ros #299: 2D backpack demo could not work.

bygreencn, 2017-04-10:
With Git repository `cartographer_ros` [319be1ad](https://github.com/googlecartographer/cartographer_ros/commit/319be1ad8cb6801bcc47fc69f00ada6d14189f36), `cartographer` [4f1914b2](https://github.com/googlecartographer/cartographer/commit/4f1914b2b46fa65f44fa752c6f100cb102554842)  and ceres-solver version 1.12.0, 2D backpack demo could not generate map, wrong status:
1. rostopic echo /tf: shows imu_link data not changing, vertical_laser_link changing with topic /imu
2. the map stop changed after first submap generated, even all topices correct read from bag file.
[image]
.

SirVer, 2017-04-10:
I am unable to reproduce this. Could you send the exact commandline you ran and maybe a screencapture of the problem you see (because the screenshot does not seem wrong to me).

bygreencn, 2017-04-10:
I just follow the comands at document about

# Download the 2D backpack example bag.
wget -P ~/Downloads https://storage.googleapis.com/cartographer-public-data/bags/backpack_2d/cartographer_paper_deutsches_museum.bag

# Launch the 2D backpack demo.
roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag

And I will show you some  screenshot Gif later. 

I found it when I do some test for 2D laser and imu, i meet some error and I have posted some screenshot about my project at [here](https://groups.google.com/forum/#!topic/google-cartographer/NaBnUT1c07Q), after that I re-test cartographer_paper_deutsches_museum.bag and I found it was almost the same.

bygreencn, 2017-04-10:
@SirVer I had got a screencapture to show you some status, please check [mp4 video file](https://drive.google.com/open?id=0B8Ugkw-ov7hqY1Z6YjllME1Td2s), I don't know whether it is correct that I learned the map coordinate system was not changed with IMU coordinate system changing.

SirVer, 2017-04-10:
@bygreencn Your problem is clearly visible in the GIF, thanks for making it.

Your terminal output shows that you are far from real-time - the sensor data is only consumed at 2-3% real time. This indicates that either you did not build in release mode (which should be impossible if you did not change the CMakeLists.txt) or that your Ceres library does not have access to all its dependencies. 

I suggest nuking your build directories and restart from scratch. Maybe your build configuration got messed up. If this does not work, I suggest trying to run inside one of our Docker containers and see if the problem is gone there.

Maybe related: https://github.com/googlecartographer/cartographer_ros/issues/247

SirVer, 2017-04-10:
Closing, as this seems to be build environment related and not really a Cartographer issue. If you think Cartographer could change something to improve the situation, I will reopen.

closed 2017-04-10 by SirVer

bygreencn, 2017-04-10:
@SirVer Yes, I run it on Debug mode, 2d demo bag file and my test projcet. Do you mean that these status is reasonable, if I changed to release version, I could get the correct result? I will try and give an response.

SirVer, 2017-04-10:
@bygreencn Yes, debug mode is 50-100 times slower, i.e. unusable. We terminate compilation for that reason when you try to compile in debug mode.