cartographer-project/cartographer_ros #304: Why the pointcloud in scan_matched_points2 is rotating

mxrchn, 2017-04-18:
I used early version (Oct 2016) of ceres_solver, cartographer, cartographer_ros.
The demo_backpack_3d.launch runs well with cartographer_3d_deutsches_museum.bag.
And I try to use the packages with my sensors. My LiDAR is velodyne VLP16, with 750 packages per sec, and the imu is razor_imu_9dof, which rate is only 50 Hz. I delete the vertical_laser_3d topic, because I have only one LiDAR. I run the demo with the bag above again, and it also goes well with only horizontal_laser_3d data.
But when I uses the launch file with my sensors, something goes wrong that the pointcloud start rotating, and submaps are hardly generated. The frame_id is map and the topic is scan_matched_points2. However, the sensors are static in the environment, and the data from the sensors are also stable.
So which parameters  should I configure? Thx a lot.

SirVer, 2017-04-18:
This is impossible to tell from your description alone. It could be that your URDF is wrong or that your topic remapping is bogus. Maybe your sensor sends bogus data? Maybe your IMU is using the wrong units?

Please provide bag, configuration files and launch files in form of a fork of `cartographer_ros`, ideally also a screen capture of your problem from rViz.

SirVer, 2017-04-24:
A hunch: Try `TRAJECTORY_BUILDER_3D.scans_per_accumulation = 1` in your lua configuration. Your node probably publishes full revolutions instead of small scans. See [this FAQ entry](https://google-cartographer-ros.readthedocs.io/en/latest/faq.html#the-laser-data-in-the-3d-bags-is-much-higher-than-the-maximum-reported-20-hz-rotation-speed-that-the-vlp-16-can-do-why) as of why the backpack 3d configuration uses something different.

closed 2017-04-24 by SirVer

JoestarK, 2017-07-26:
@mxrchn Hi,
I got the same problem as yours. Did you solve it ?

ytgcljj, 2018-01-22:
I got the same problem as yours. Did you solve it ?

Conhokis, 2019-04-22:
Had the same problem, turns out it was an IMU problem where the gravity value 10x under the supposed one.