cartographer-project/cartographer_ros #305: Using Cartographer as odometry with amcl for localization in a given map

oya123, 2017-04-18:
I wanted to exploit the fact that Cartographer gives pretty accurate position estimate. I used Cartographer just using IMU and laser to make an indoor map, and it pretty good. 

Then I want use Cartographer as odometry and amcl to do global localization in a pre-aquired map by Cartographer, but the robot drift away.

On my robot a TF frame that looks like this:
_**map-->odom**_: comes from /amcl

_**odom-->google_odom**_: comes from /cartographer_node

**_google_odom-->base_footprint_**: comes from /cartographer_node

_**base_footprint --> base_link**_: comes from /robot_state_publisher

**_base_link --> gryo_link_**: comes  from /robot_state_publisher

**_base_link --> laser_frame_**: comes  from /robot_state_publisher

The configuration of Cartographer_ros is like:

```
include "map_builder.lua"

options
 ={
  map_builder = MAP_BUILDER,
  map_frame = "odom",
  tracking_frame = "gyro_link",
  published_frame = "base_footprint",
  odom_frame = "google_odom",
  provide_odom_frame = true,
  use_odometry = false,
  use_laser_scan = true,
  use_multi_echo_laser_scan = false,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.laser_min_range = 0.2
TRAJECTORY_BUILDER_2D.laser_max_range = 8.
TRAJECTORY_BUILDER_2D.laser_missing_echo_ray_length = 5.
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)

SPARSE_POSE_GRAPH.constraint_builder.min_score = 0.65
SPARSE_POSE_GRAPH.constraint_builder.global_localization_min_score = 0.7

return options
```
I used the position estimate to find odometry and now Cartographer successful transform the required tf. But when I run amcl and publish an already created map using map_server, the pose of robot drift. It seems amcl doesn't work enven if the tf, map, laser_scan look like ok. And I also remap the Cartographer mapping map to something else . Please help how to resolve this issue.

SirVer, 2017-04-18:
Cartographer does not really work as pure localization right now. TRI is interested in building this functionality in the mid term though.

Paging @macmason .

macmason, 2017-04-19:
Yes, TRI (me et al.) are thinking about pure localization, but don't hold your breath. We're headed there by way of lifelong SLAM, because lifelong is more important for us, so it's a ways off.

I've had good luck in the past by pulling out just the scan matcher and using them for "laser odometry" when I'm not doing SLAM. Your mileage may vary.

HolmarSeffelgruber, 2017-04-20:
What do you mean by pulling out the scan matcher? Are you still using cartographer? The scan matcher is using the submap I guess. I'm trying a mcl localization right now, but it is not nearly as good as cartographer. I'm using 3d cartographer though.

macmason, 2017-04-22:
I've used the scan matchers (without the rest of cartographer) to do scan-to-scan matching to generate "odometry" which I then fed into AMCL. If your environment has some structure, that does a pretty good job of tracking the robot; better than wheels in many cases.

HolmarSeffelgruber, 2017-04-24:
Would you mind giving me the code how you got and provided the odometry? I'm not sure if  I'm able to do it myself since I am not a programmer throughout but I might be able to adjust it for my 3d purpose. My environment is good enough for cartographer to provide pretty good maps without odometry so I think the structure is good for the scan matcher. However the MCL localization has big trouble with movements and we'd like to be independent from wheel odometry.

macmason, 2017-04-24:
Sadly, I don't have access to that code anymore. The basic idea was to use `RealTimeCorrelativeScanMatcher` (which is in `mapping_2d/scan_matching`). Suppose you have two scans, `a` and `b` (where `b` is later in time). Then you'd render `a` into a `ProbabilityGrid`, and then call the `Match` method to get a good `pose_estimate`.

In my empirical testing, this was so much better than my wheel odometry (for a differential-drive robot with very flexible tires) that I just ignored the wheels entirely.

damonkohler, 2017-05-12:
Closing due to inactivity. Pure localization is on the near term roadmap, but not yet supported.

closed 2017-05-12 by damonkohler