cartographer-project/cartographer_ros #307: hybrid grid in 3D cartographer

sunmk2006, 2017-04-19:
whether it is 3D grid map in 3D cartographer about the hybrid grid of submap? and scan matching occurs between the 3d laser fan and the 3d grid?but why do you display a 2d submap list in rviz subcribe a service of submap_query?

SirVer, 2017-04-19:
2D is easier to understand and make sense of than 3D voxels. So we flatten the submap into a single texture (akin to the X-Ray visualization) and ship that to rViz.

closed 2017-04-19 by SirVer

sunmk2006, 2017-04-20:
Can I understand that the existence of 3D grids is the need for scan matching methods in 3D cartographer,and 2D grid is for 2D laser scan matching, not for other reasons?

SirVer, 2017-04-20:
I do not fully parse the question.

> Can I understand that the existence of 3D grids is the need for scan matching methods in 3D cartographer

that is definitively correct. 

> ,and 2D grid is for 2D laser scan matching

that is also correct.

> not for other reasons

Is this part of the question if the HybridGrid and ProbabilityGrid are used for other stuff? If so, the HybridGrid is used in X-Ray generation as well, maybe other places. But it is mostly for scan matching.