cartographer-project/cartographer_ros #308: separating rviz_plugin from cartographer_ros

jihoonl, 2017-04-19:
`cartographer_rviz` brings too big dependencies to install in a robot. Could we separate this package out as an independent repo called like `cartographer_ros_desktop`

SirVer, 2017-04-19:
That is a valid request. This is theoretically simple too, but it is a rather big refactoring that we do not want to tackle right now. We gladly review PRs of course.

jihoonl, 2017-04-19:
Can you create a new repo for `cartographer_rviz`? Then I will try to create PRs for refactoring

damonkohler, 2017-04-19:
Done. There's now a cartographer_rviz repo.

SirVer, 2017-06-08:
I think this just requires an FAQ entry now and can then be closed.

SirVer, 2017-10-12:
In offline discussions we decided that the options ROS offers for excluding packages from building are sufficient to tackle this. #515 added a FAQ entry to explain how.

closed 2017-10-12 by SirVer