cartographer-project/cartographer_ros #310: Clock synchronization

sunmk2006, 2017-04-24:
How to deal with the clock synchronization between sensors（for example imu and lidar) in cartographer?

SirVer, 2017-04-24:
How do you envision Cartographer help here? 

I think, it must be done externally. Ideally by using a common clock (like GPS or a custom timing board) or triggering.

closed 2017-04-24 by SirVer

bygreencn, 2017-05-10:
Hi @SirVer:
I also had this question. As we known, when Sendor driver packages grad the sendor data, generally it adds ROS System's time as timestamp into ROS topic message, then published it  to cartographer. Mostly sensors do not have external clock synchronization, such as PPS to trigger these sensors. And also if we use IMU, Lidar and without Odom, we want use cartographer published odom to do some posted operation, how to know accurate position of time and location from Odom?

**USE ?**
1. ROS2;
2. Make ROS as a real-time operating system and add the real-time layer.

**BUT**
When the computer load is heavy, how to guarantee the clock synchronization, do you have any operable suggestions? Hope time precision as small as possible, such as 20ms.
Thanks
BR
bygreencn

SirVer, 2017-05-10:
>  Mostly sensors do not have external clock synchronization...

This is not my experience. Professional grade sensors usually allow for some sort of time synchronization. Quite often the exchange format is GPS timestamps (which can be faked from a common source of timing information). This is the only reliable way of time syncing your data.

> Hope time precision as small as possible, such as 20ms.

That is not a high bar - you can easily achieve that just by using system time and providing each sensor with a separate USB bus or NIC to use. For some of our cases it is also not sufficient: A car going 180 km/h travels 44cm in that time frame - a way too low accuracy for SLAM.