cartographer-project/cartographer_ros #316: Error when building according to https://google-cartographer-ros.readthedocs.io/en/latest/

ThaHypnotoad, 2017-04-26:
ubuntu 16.04 
ROS Kinetic

Ran the commands in the Building & Installation section and got a cmake error when it finished with ceres-solver and moved on to cartographer. The full gist is [here
](https://gist.github.com/ThaHypnotoad/e112a93362f988c3d02e73e325612a1e)

I can't seem to get the formatting to work here, but the relevant bit is right here:

 `
\-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
CMake Error at /home/aleks/Documents/catkin_ws/install_isolated/lib/cmake/Ceres/CeresConfig.cmake:88 (message):
  Failed to find Ceres - Missing requested Ceres components:
  [SparseLinearAlgebraLibrary] (components requested:
  [SparseLinearAlgebraLibrary]).  Detected Ceres version: 1.12.0 installed
  in: /home/aleks/Documents/catkin_ws/install_isolated with components:
  [LAPACK, SchurSpecializations, OpenMP].
`

I have Eigen and SuiteSparse installed, but it thinks that neither is available? Or at the very least it's finding Eigen and shouldn't care whether or not I have SuiteSparse since Ceres should be able to use Eigen in its place. I'm not entirely sure what to make of this error since this is the first time I've seen an error of this type.

ThaHypnotoad, 2017-04-26:
Update: Seems to work fine without ninja. Wonder what it's doing so differently. (I've never used ninja before)

SirVer, 2017-04-26:
Your gist does not seem to be from the first running of cmake - could you wipe your workspace, start over and provide the full output? Also, did you try building inside a fresh docker image of ubuntu 16.04 to rule out something weird in your environment? I consider this to be the most likely explanation to your problem so far, since we have reported successes on 16.04 + kinetic.

> Seems to work fine without ninja.

That is very unlikely - ninja is just a build system, it does not change the build logic. It is faster than GNU make, but otherwise very similar. It is more likely that make just ran the build, ignoring the configuration errors. This will work, but Cartographer will be roughly 2 orders of magnitudes slower since Ceres will not have a sparse matrix library.

ThaHypnotoad, 2017-04-26:
Hmm very odd. It seems to have gone off without a hitch this time (the fourth time I've tried). I'm going to see if I can reproduce the error later tonight.

ThaHypnotoad, 2017-04-27:
Alright I'm having trouble reproducing the error. Is there a state we can set this to that reflects that or shoudl I just close it and reopen it later if and when I can reliably reproduce the issue?

damonkohler, 2017-04-27:
We'll close it as not reproducible. You can still followup on this issue if you find you're able to reproduce it in the future.

closed 2017-04-27 by damonkohler

hcv1027, 2017-06-13:
Hi, I just have the same issue.
My environment is Ubuntu 16.04 + ROS Kinetic.
What information do I need to provide?

hcv1027, 2017-06-13:
I just create a new empty folder as the working space and follow the build/install step again. This time it worked completely without any error. Seem below steps will not complete because other ros packages in my original ros working space folder.
```
sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y 
```

SirVer, 2017-06-13:
@hcv1027 The way you describe it, it sounds like a ROS/catkin/rosdep issue. Can you reproduce this reliably? If so, it would be worthwhile to file a bug with rosdep.

hcv1027, 2017-06-14:
@SirVer I will try it. There are multiple packages in my original ros working space. Not sure which one cause this problem.

ixgn, 2017-12-06:
I had the same issue. Removing the setup of any other custom packages from my .bashrc and recompiling without ninja did the trick.