cartographer-project/cartographer_ros #317: can't make the good result using real time processing data

carl1218, 2017-04-27:
Hi.
  when i run the cartographer_ros using real time data,can't make the good result , if the size of scene not large,the result seems  good,But when the scene is a little bigger,the result seems not good,especially when i Return along the original path,The map is a mess,the map look like this:
[image]

this is launch and lau:
https://drive.google.com/file/d/0B33M6O52cUeNM2NhVWFwRmhQVzA/view?usp=sharing
https://drive.google.com/file/d/0B33M6O52cUeNTkZPUVhZWEYwQm8/view?usp=sharing

this is bag file:
https://drive.google.com/file/d/0B33M6O52cUeNbEREVk1IaFdwZHM/view?usp=sharing

SirVer, 2017-04-28:
Your example has two profound problems: 

1) The outdoor environment violates the 2D assumption (flat floor, perpendicular walls, laser is perpendicular to the ground). This means that you will not be able to get great results there ever.
2) Your laser timestamps are wonky. Your laser is 40Hz, so you'd expect consecutive timestamps to be 25ms apart. This is true for your bag on average, but the smallest value 0.44ms. Since these differences in timing will be used for velocity estimation, the velocity predictions will go through the roof - this is the sliding you are seeing.

My suggestion is to fix your data - for example by hard coding that each scan is 25ms after the former instead of using your system clock for timing the scans. This should allow you to get a reasonable map. I opened a bug about the poor resilience of velocity estimation. https://github.com/googlecartographer/cartographer/issues/242

closed 2017-04-28 by SirVer

SirVer, 2017-04-28:
Another thought: Switching to http://wiki.ros.org/urg_node instead of the hokuyo node gives you more control over the timing. See `calibrate_time` and `time_offset`.

carl1218, 2017-04-29:
@SirVer ,Thank you very much.