cartographer-project/cartographer_ros #318: How does it work with TUM RGB-D SLAM Dataset

zhehangT, 2017-04-28:
Thank you for excellent job.

I am very interested at how's cartographer going to perform on TUM RGB-D SLAM Dataset .
But I failed to get any map when I run cartographer and TUM RGB-D SLAM Dataset rosbag file.
It seems something wrong with the TF Transformation, but I am not sure.
The rosbag file link is http://vision.in.tum.de/data/datasets/rgbd-dataset.

Looking forward to your favourable reply.

SirVer, 2017-07-31:
I am sorry, I will not be able to look into this for the next 2 months. I am very interested how Cartographer performans and expect that it should work well.

janboeye, 2017-08-18:
I made following changes and try to start it, 
diff --git a/cartographer_ros/configuration_files/tum_rgbd.lua b/cartographer_ros/configuration_files/tum_rgbd.lua
index f12945f..22da09e 100644
--- a/cartographer_ros/configuration_files/tum_rgbd.lua
+++ b/cartographer_ros/configuration_files/tum_rgbd.lua
@@ -13,9 +13,11 @@
 -- limitations under the License.
 
 include "map_builder.lua"
+include "trajectory_builder.lua"
 
 options = {
   map_builder = MAP_BUILDER,
+  trajectory_builder =  TRAJECTORY_BUILDER,
   map_frame = "map",
   tracking_frame = "laser",
   published_frame = "odom",
@@ -23,11 +25,14 @@ options = {
   provide_odom_frame = false,
   use_odometry = true,
   use_laser_scan = true,
-  use_multi_echo_laser_scan = false,
+  num_laser_scans = 1,
+  num_multi_echo_laser_scans = 0,
+  num_subdivisions_per_laser_scan = 0,
   num_point_clouds = 0,
   lookup_transform_timeout_sec = 0.2,
   submap_publish_period_sec = 0.3,
   pose_publish_period_sec = 5e-3,
+  trajectory_publish_period_se = 30e-3,
 }
 
 MAP_BUILDER.use_trajectory_builder_2d = true

But it reports laser_min_range was used the wrong number of times and then crash.

janboeye, 2017-08-18:
I changed tum_rgbd.lua as following, then I could start it, but get same result as what you get. 
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder =  TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "laser",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  use_odometry = true,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.min_range = 0.1
TRAJECTORY_BUILDER_2D.max_range = 4.0
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.motion_filter.max_time_seconds = 0.3
TRAJECTORY_BUILDER_2D.motion_filter.max_distance_meters = 0.2
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(3.)
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight = 70
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight = 300

TRAJECTORY_BUILDER_2D.submaps.resolution = 0.035
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 120
SPARSE_POSE_GRAPH.optimize_every_n_scans = 120
SPARSE_POSE_GRAPH.constraint_builder.min_score = 0.82
SPARSE_POSE_GRAPH.constraint_builder.sampling_ratio = 1.

return options

janboeye, 2017-08-18:
it looks like that it is because the bag file's depth image is 32fc1 format, but depthimage_to_laserscan does not support this. @zhehangT

bochen87, 2020-10-16:
closing, as inactive for a long time

closed 2020-10-16 by bochen87