cartographer-project/cartographer_ros #319: lookuptable and probabilitygrid in  cartographer

sunmk2006, 2017-05-02:
To get the value of the probability grid, cartographer looks up a table with hit_table_ or miss_table_,but why does the table correspond to the probability of the grid?

wohe, 2017-05-02:
The mapping between (integer) values in the probability grid and the probabilities they represent is done by `ProbabilityToValue` and `ValueToProbability`.

`hit_table_` and `miss_table_` are used to update the value in a single pixel or voxel. It maps from a pixel (or voxel) value to the new value after update. This new value includes setting the update marker (the most significant bit). This is all done with a single lookup for efficiency.

Does this answer your question?

sunmk2006, 2017-05-02:
How many grid points a pixel contains?The number of hit points in a pixel indicate the hit probability,and the miss points indicate the miss probability, right?

wohe, 2017-05-02:
Each pixel corresponds to one grid point.

What exactly do you mean with hit probability? The parameter of the same name? This gets used to update the value of a pixel when there was a hit.

sunmk2006, 2017-05-02:
It has a variable in the file of ./cartographer/mapping_2d/ray_casting.cc 
//Factor for subpixel accuracy of start and end point.
constexpr int kSubpixelScale = 1000;
Does it mean the number of grid points a pixel contains?

wohe, 2017-05-02:
No. As I said there is exactly one pixel per grid point.

The parameter `kSubpixelScale` you mention is part of the ray casting algorithm used in 2D SLAM. It has nothing to do with probabilities, pixel size, etc. The range data input to `CastRays()` is using floating point. These values, i.e. line segment end points, get discretized to 1/1000 (this is determined by `kSubpixelScale`) of a pixel. Then all pixels are determined which get hit by these line segments.

sunmk2006, 2017-05-03:
From scan points to grid point or pixel, many scan points fall into a pixel,the probability of the grid is dynamic and will be updated many times，in other words,the probability of the grid is proportional to the number of times the grid is queried,right?

wohe, 2017-05-03:
This depends on where the multiple scan points come from:
1. For multiple scan points of a single scan that fall into the same pixel, this is not the case. For each scan that is inserted, each pixel gets updated at most once.
2. For multiple scans that are inserted and see the same pixel as obstructed or free, the probability of this pixel will increase or decrease with each update.

sunmk2006, 2017-05-04:
OK,but there is another quesstion,if the probability of a pixel has two different attributes hit probability and miss probability sametimes(P|(p_hit,p_miss)) or it only has a value(P= P_hit - P_miss),I can understand their respective increase from their loopuptable(P=p_hit+p_hit+... , P=p_miss+p_miss+...),but How does its decrease  happen with each update?

wohe, 2017-05-04:
No, this is not how updates work. You should probably read https://research.google.com/pubs/pub45466.html where this is described in detail. Closing this issue as it is not actionable.

closed 2017-05-04 by wohe