cartographer-project/cartographer_ros #330: Does Cartographer publish an occupancy map in 3D mode?

msy22, 2017-05-08:
The [cartographer_ros docs](https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html) state that the published occupancy map is "only supported in 2D". To clarify, does this mean:

1. The 2D occupancy map will only be published if Cartographer is in 3D mode. --**OR**--
2. The 2D occupancy map is always published, regardless of mode, and the statement simply refers to the fact that the occupancy map is 2D and not 3D.

Thanks in advance for any help on the issue.

damonkohler, 2017-05-09:
3. The 2D nav_msgs/OccupancyGrid is published only when Cartographer is in 2D mode. Only x-ray submaps are published in 3D mode.

msy22, 2017-05-09:
Thanks for your help, but now I have some follow-up questions:

1. What do you mean by x-ray submaps?
2. If no occupancy grid is published, then how does a robot using Cartographer know where obstacles are?

SirVer, 2017-05-10:
1. ​Damon means our visualization of 3D submaps if you run 3D SLAM. In rViz we show a projected down 2D plane of the 3D submap data and we call these x-rays.​

2. The short answer here is that they do not: No code has been written that can navigate or reason over Cartographer's submaps representation (yet) for trajectory planing and collision avoidance.

The occupancy grid publishing is a hack around for people that require a live map for a quick test or demo. It is a hack though, since occupancy grids cannot scale to the size of problems Cartographer can handle and do not efficiently represent the dynamic nature of environments. For 3D we do not provide such a hack at all - it could theoretically be added. In the midterm we plan to move the Cartographer submaps -> ROS occupancy grid conversion into a separate node and probably move it out of Cartographer. 

In the short term, you are probably better off by running your robot around and SLAM your environment, write out the trajectory and create a map (rosservice call /finish_trajectory) and serve this up using mapserver.

msy22, 2017-05-12:
So in 3D mode, the submap (3D) is projected onto a plane (2D) using x-rays. Is this still published on the `map` topic with the `nav_msgs/OccupancyGrid` format? If so, and this is a usable occupancy map, couldn't it be fed to the ROS navigation stack (specifically the `move_base` node) for navigation purposes?

damonkohler, 2017-05-12:
X-rays are not published to the /map topic as nav_msgs/OccupancyGrid. The /map topic is only supported in 2D.

Publishing nav_msgs/OccupancyGrid very inefficient for large space. Additionally, x-rays would likely not make useful occupancy grids for 2D navigation since they include all structure projected into a 2D plane. X-rays are a 2D representation of vertical density that's useful for humans, but probably not for robots.

msy22, 2017-05-15:
So from what I understand, Cartographer in it's current form is better suited to mapping than autonomous navigation in unknown environments. Is that correct?

Am I then correct in assuming that when collecting the PR2 and Taurob Tracker datasets, the robots were manually driven? or did you do as SirVer suggested above and have a pre-prepared map to use for navigation? 

Thanks again for your help and patience with me trying to understand Cartographer.

damonkohler, 2017-05-15:
For 3D, yes, that's correct. In 2D, everything is already in place for autonomous navigation.

The PR2 datasets were collected long before Cartographer existed. I believe they were generated autonomously. The Taurob Tracker was manually driven AFAIK.

closed 2017-05-15 by damonkohler

SirVer, 2017-05-15:
>  it's current form is better suited to mapping than autonomous navigation in unknown environments

This is definitively correct, since there are no path planers (yet) that work over the Cartographer submap representation. I argue though that publishing one huge `map` (like ROS does and its tools expect) is a badly scaling approach. Should a planer be build on top of cartographer submaps (or similar concepts) the algorithms would work better better with respect to scale, data freshness and path finding.

msy22, 2017-05-17:
Ok, thanks for the help and the clarification!