cartographer-project/cartographer_ros #333: SLAM with 180 degree laser scans

CyanChiu, 2017-05-10:
Dear cartographer ros team,
Very thanks for your maintenance of this open source SLAM, and make programmers can learn this domain easier.

I have a problem that I have to use the 180 degree laser scan on my robot, and this situation make the cartographer build a poor map as below image. Would you recommend me how to fine tune the configurations or if I can improve it by anything(add weight of odom...)?
[poor image](https://drive.google.com/open?id=0B3wI8JrYusSeZFkzdktILTY2ODg)

[bag file and config](https://drive.google.com/open?id=0B3wI8JrYusSeNk5mT3FaQ2h0OTQ)
[video](https://youtu.be/JL90_5fL4OY)

```
options = {
  map_builder = MAP_BUILDER,
  map_frame = "map",
  tracking_frame = "base_footprint",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  use_odometry = true,
  use_laser_scan = true,
  use_multi_echo_laser_scan = false,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 1,
  pose_publish_period_sec = 0.033,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.laser_min_range = 0.48
TRAJECTORY_BUILDER_2D.laser_max_range = 5.5
TRAJECTORY_BUILDER_2D.laser_missing_echo_ray_length = 5.
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)

SPARSE_POSE_GRAPH.optimization_problem.huber_scale = 1e2

return options
```

SirVer, 2017-05-10:
The video was quite helpful. It shows that your odometry is rather poor when you drive downwards - you can see how bent the long hallway gets. Once you turn the corner your rooms look similar to above and Cartographer inserts a wrong loop closure for this.

You should probably increase SPARSE_POSE_GRAPH.constraint_builder.min_score to accept only better loop closures and also maybe decrease the search window for loop closures (also in SPARSE_POSE_GRAPH). I'd expect that once you drive the same way back Cartographer figures out that your odometry is biased towards a right curve and fixes the map.

I did not investigate your data yet, let me know if this does not get you to your desired result and I can investigate your data.

CyanChiu, 2017-05-12:
Sorry for the late reply and thanks for your recommend.

As you said, I want to remove the wrong matching at long hallway and add the min_score config  as attached below.
The hallway is not bent this time, but maybe it is too similar to make robot not go ahead in the map.
(Video 2:40). For the search window size, I don't know which config to fine tune it and take time to study the spare pose graph and paper now. I will try other config value latter and if you can give me more suggestion I will try it:)
[Video](https://drive.google.com/open?id=0B3wI8JrYusSea01QYi11S1htalE)
[bag(laser and odom without slam) and config file](https://drive.google.com/open?id=0B3wI8JrYusSeQlpUZDFyRnNVY2s)

```
options = {
  map_builder = MAP_BUILDER,
  map_frame = "map",
  tracking_frame = "base_footprint",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  use_odometry = true,
  use_laser_scan = true,
  use_multi_echo_laser_scan = false,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 1,
  pose_publish_period_sec = 0.033,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.laser_min_range = 0.48
TRAJECTORY_BUILDER_2D.laser_max_range = 5.5
TRAJECTORY_BUILDER_2D.laser_missing_echo_ray_length = 5.
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)

MAP_BUILDER.sparse_pose_graph.optimization_problem.huber_scale = 1e2
MAP_BUILDER.sparse_pose_graph.constraint_builder.min_score = 0.7
return options
```

SirVer, 2017-05-15:
I did some quick tuning over your data. Thanks for providing all the information!

You can find all significant changes in [this commit](https://github.com/SirVer/cartographer_ros/commit/1b271234a98ab5882c0f7624355df3290f700cc7). It created the map below.

The important changes are reducing `num_range_data` to make your submaps smaller, so that small drifts do not destroy too much of the map, increasing the `min_score` to get fewer and more meaningful loop closing constraints and cranking up the `translation_weight` in local SLAM. This trusts the current velocity estimate more which reduces slipping when the scan matching fails in hallways.

I also think that your laser mount is tilted, i.e. your laser is not perfectly perpendicular to the ground plane. You can see this when your robot drives one way turns around and back again: the hallways disagree, always turning slightly upwards. You should check your URDF and/or your laser mounts.

[image]

closed 2017-05-15 by SirVer

CyanChiu, 2017-05-16:
It's awesome! Thanks for your great help and suggestion, I will try more parameters.