cartographer-project/cartographer_ros #334: Detecting Loop Closure & Resolution

CycleMark, 2017-05-10:
Hi,

I've got cartographer working on my robot with and RPLIDAR A2 basically using the turtlebot_urg_lidar_2d.lua config.

Couple of questions I have:
1) How/what topic can I listen to to determine loop closure?
2) The resolution (in rviz) appears to be relatively coarse compared to gmapping. What config(s) can I change to alter this?

Many Thanks

Mark

SirVer, 2017-05-11:
1. The constraints are currently not published, but there is https://github.com/googlecartographer/cartographer_ros/pull/325 which makes them available to rviz. 

2. The settings in [TRAJECTORY_BUILDER_2D.submaps](https://github.com/googlecartographer/cartographer/blob/master/configuration_files/trajectory_builder_2d.lua). Do not change them in this file directly, instead overwrite them in your .lua file. Changing this setting will probably mean that you have to retune loop closure and local SLAM settings for good performance though.

The default resolution for the submaps is 5cm - it is close the physical limit reachable of hokuyo lasers (I never got better than 3cm with them). I doubt that gmapping is giving a higher resolution, maybe they smooth their visualization? Screenshots of what concerns you would be helpful to give you a better answer.

closed 2017-05-11 by SirVer

CycleMark, 2017-05-11:
Hi,
I've attached a couple of screen shots. Should help show what I mean.
Thanks
Mark
[image]
[image]

SirVer, 2017-05-12:
Sorry, the screenshots do not help me.

Both seems to have a similar resolution and both maps sorta look fine. None of them visualize Cartographer's submaps so I cannot tell which one is from Cartographer and which one is from gmapping.The top one has the laser fan in the wrong position - my initial guess would be two nodes publishing to the odom topic. Have you run roswtf?

Could you point out what you want to convey with the screenshots?

CycleMark, 2017-05-13:
HI Sorry - don't worry about the laser - its actually fine (I was just getting a screen shot for reference)
The top one is gmappping the bottom one Cartographer. Both of more or less the same environment - rotate the bottom 90 deg clockwise and they'd be the same.

My question was the cartographer one looks less sharp than the gmapping one

SirVer, 2017-05-15:
@CycleMark I think what you are referring to is due to the cutoff probability. Cartographer uses a continious representation, while ROS' occupancy grid expects a ternary representation - so in the conversion you might get this unsharpeness. 

You can tune these parameters or (recommended) use Cartographer's visualization which contains more information anyways.