cartographer-project/cartographer_ros #336: Reliability evaluation of the output trajectory

sunmk2006, 2017-05-12:
The output trajectory is primary function of cartographer,but what extent can I believe this trajectory?Are there some parameters in cartographer or algorithms that can evaluate the reliability of trajectories?
What's your suggestion?

damonkohler, 2017-05-12:
Are you asking for something like covariance? Or are you asking how you can evaluate the quality of the trajectory compared to something like ground truth?

sunmk2006, 2017-05-15:
Something like that，but how do we evaluate the quality of the trajectory without ground truth?
for example, does the matching algorithm or optimization algorithm have some output as a reference?

damonkohler, 2017-05-15:
No, I don't think there's anything like that available. Evaluating the quality of a trajectory beyond subjective assessment of the x-rays is not something we developed a satisfactory strategy for.

closed 2017-05-15 by damonkohler

sunmk2006, 2017-05-15:
What  x-rays have you mentioned many times?

SirVer, 2017-05-15:
There is some ground truth data sets that can be used to judge the performance of Cartographer. See our paper about 2D examples. There is also work to look at the performance of Cartographer in googlecartographer/cartographer_ros#332.

sunmk2006, 2017-05-22:
I still want to get cartogrpher's own evalution for some reason，even if it doesn't look so accurate or unwise,do you have any suggestions，i would appreciate it very much.

SirVer, 2017-05-26:
Cartographer does not provide an evaluation of its own output - it is not a fully probabilistic system that gauges the estimate of its quality. The only thing you can do is compare it against some know ground truth. Our paper contains benchmarks against ground truth benchmarks for 2D and Cartographer fairs very well there. 

In general, given enough loop closure and proper tuning, 2D SLAM error is around the voxel size. But this can never be generalized. For example looking at googlecartographer/cartographer_ros#332 you can see parts of buildings connected by hallways. There are no true loops, i.e. alternative ways to get from the left most wing to the right most wing - only one hallway. This means that slipping in the corridor or slight angle errors can lead to large global errors in these wings. While locally the SLAM will still be ~5cm accurate, the global position error (i.e. the lat/long of the walls in the wings) will be larger.

We never compared 3D SLAM against public datasets and published results. From my experience it works as well as 2D, but I have no hard data to proof this.

sunmk2006, 2017-05-26:
Do you have some data that  know ground truth for cartographer 3D model，or some public data sets, I  want to do some tests on accuracy.

SirVer, 2017-05-26:
No, as I said we did not do this for 3D yet. Should we ever get around to writing a paper about 3D, we will start investigating. 

If you find any and do an investigation, please let us know, we are very interested in the results.

sunmk2006, 2017-05-27:
It's my pleasure to help you.
What's more, I don't quite understand the accuracy of the trajectory test.The precision of the tracking(2cm accurate or 5cm accurate) is how to measure out,Is it the total length error of the trajectory or the average pose error of each point on the trajectory?
Can I improve the trajectory accuracy if I adjust that parameter(the voxel size)?

SirVer, 2017-05-29:
The accuracy metric (comparing to ground truth) is described in our paper.

I answered your last bullet point [here as answer to bullet 5](https://github.com/googlecartographer/cartographer_ros/issues/351#issuecomment-304266232).

sunmk2006, 2017-06-01:
hi,do you know [the KITTI odometry dataset](http://www.cvlibs.net/datasets/kitti/eval_odometry.php) , i want to use it to do 3D cartographer accuracy tests , A few problems come up ：

1. the algorithm needs to use IMU information, but the IMU information is not found in this dataset.

2. the point clouds contain a sensor_msgs/PointCloud2 and timestamp per revolution,not one per UDP packet,Is it necessary to slice it and how to slice it? 

3. After slicing, how does the output trajectory correspond to the ground truth in the dataset?

SirVer, 2017-06-01:
Yes, I know KITTI - I was at the same institute as Andreas Geiger & Philip Lenz when they created Kitti and they are close friends.

See also #248 where @kentsommer looked into Kitti already.

1. IMU is provided in the raw data: http://www.cvlibs.net/datasets/kitti/raw_data.php
2. You have to write code for doing this. There are kitti raw -> ROS converters that are easy to adapt.
3. You have to find a rigid transform that minimizes the trajectory error. This is your frame conversion between kitti origin and cartographer origin. The remaining error is the Cartographer SLAM error. The kitti paper contains information how Andreas & Philip decided to compare trajectories and there is sample code on the kitti site to do it.

sunmk2006, 2017-06-01:
For a point cloud per revolution,  if i don't know the order in which the sensor receives the point cloud, how to slice, and give it  a time tag like a UDP packet.

SirVer, 2017-06-01:
The raw data contains the points in order. The firing pattern of the Velodyne is documented in its data sheet.

sunmk2006, 2017-06-13:
From the document of Velodyne, I think that the time needed to send an UDP package should be equal，for example,vlp16 need 1306.37μs，HDL32E need 552.96μs,HDL64E S2 288μs and so on. i test it using a bag named b3-2016-04-05-15-51-36.bag from [here](http://google-cartographer-ros.readthedocs.io/en/latest/data.html),diff = this_stamp - last_stamp, Some examples are shown below, the values are not equal , Why is that? 
[ INFO] [1497251736.491805168]: 3123: this_stamp:1459871506.394222 diff:0.000664
[ INFO] [1497251736.492418176]: 3124: this_stamp:1459871506.394885 diff:0.000663
[ INFO] [1497251736.499990479]: 3125: this_stamp:1459871506.396877 diff:0.001992
[ INFO] [1497251736.500034565]: 3126: this_stamp:1459871506.402379 diff:0.005502
[ INFO] [1497251736.500438202]: 3127: this_stamp:1459871506.402870 diff:0.000491
[ INFO] [1497251736.501044527]: 3128: this_stamp:1459871506.403516 diff:0.000646
[ INFO] [1497251736.508436107]: 3129: this_stamp:1459871506.404821 diff:0.001305
[ INFO] [1497251736.508479246]: 3130: this_stamp:1459871506.410869 diff:0.006048
[ INFO] [1497251736.509638672]: 3131: this_stamp:1459871506.411503 diff:0.000633
[ INFO] [1497251736.509687734]: 3132: this_stamp:1459871506.412164 diff:0.000662
[ INFO] [1497251736.517059439]: 3133: this_stamp:1459871506.413503 diff:0.001338
[ INFO] [1497251736.517104800]: 3134: this_stamp:1459871506.419488 diff:0.005985
[ INFO] [1497251736.517695575]: 3135: this_stamp:1459871506.420136 diff:0.000648
[ INFO] [1497251736.518374889]: 3136: this_stamp:1459871506.420818 diff:0.000682
[ INFO] [1497251736.525784931]: 3137: this_stamp:1459871506.422812 diff:0.001994
[ INFO] [1497251736.525832005]: 3138: this_stamp:1459871506.428072 diff:0.005260
[ INFO] [1497251736.526288119]: 3139: this_stamp:1459871506.428758 diff:0.000686
[ INFO] [1497251736.526935430]: 3140: this_stamp:1459871506.429393 diff:0.000634
[ INFO] [1497251736.534261017]: 3141: this_stamp:1459871506.433341 diff:0.003948
[ INFO] [1497251736.534293625]: 3142: this_stamp:1459871506.436690 diff:0.003349
[ INFO] [1497251736.534947532]: 3143: this_stamp:1459871506.437352 diff:0.000662
[ INFO] [1497251736.535544087]: 3144: this_stamp:1459871506.438016 diff:0.000664
[ INFO] [1497251736.542872404]: 3145: this_stamp:1459871506.439343 diff:0.001327
[ INFO] [1497251736.542916230]: 3146: this_stamp:1459871506.445316 diff:0.005972
[ INFO] [1497251736.543537391]: 3147: this_stamp:1459871506.445990 diff:0.000675
[ INFO] [1497251736.544158731]: 3148: this_stamp:1459871506.446650 diff:0.000660
[ INFO] [1497251736.551616199]: 3149: this_stamp:1459871506.448658 diff:0.002008
[ INFO] [1497251736.551662352]: 3150: this_stamp:1459871506.453943 diff:0.005285
[ INFO] [1497251736.552201729]: 3151: this_stamp:1459871506.454576 diff:0.000633
[ INFO] [1497251736.553010516]: 3152: this_stamp:1459871506.455467 diff:0.000891
[ INFO] [1497251736.560261664]: 3153: this_stamp:1459871506.460056 diff:0.004590
[ INFO] [1497251736.560301129]: 3154: this_stamp:1459871506.462668 diff:0.002612
[ INFO] [1497251736.560914215]: 3155: this_stamp:1459871506.463367 diff:0.000699
[ INFO] [1497251736.561585453]: 3156: this_stamp:1459871506.464046 diff:0.000678
[ INFO] [1497251736.568857004]: 3157: this_stamp:1459871506.466031 diff:0.001986
[ INFO] [1497251736.568887164]: 3158: this_stamp:1459871506.471313 diff:0.005282
[ INFO] [1497251736.569521994]: 3159: this_stamp:1459871506.471979 diff:0.000666
[ INFO] [1497251736.570178389]: 3160: this_stamp:1459871506.472642 diff:0.000663
[ INFO] [1497251736.577465921]: 3161: this_stamp:1459871506.474630 diff:0.001988
[ INFO] [1497251736.577495107]: 3162: this_stamp:1459871506.479923 diff:0.005293
[ INFO] [1497251736.578129214]: 3163: this_stamp:1459871506.480587 diff:0.000664
[ INFO] [1497251736.578791235]: 3164: this_stamp:1459871506.481252 diff:0.000665
[ INFO] [1497251736.586153565]: 3165: this_stamp:1459871506.483241 diff:0.001988
[ INFO] [1497251736.586205214]: 3166: this_stamp:1459871506.488550 diff:0.005310
[ INFO] [1497251736.586778234]: 3167: this_stamp:1459871506.489213 diff:0.000662
[ INFO] [1497251736.587428651]: 3168: this_stamp:1459871506.489876 diff:0.000663
[ INFO] [1497251736.594742174]: 3169: this_stamp:1459871506.493196 diff:0.003320
[ INFO] [1497251736.594781030]: 3170: this_stamp:1459871506.497177 diff:0.003981
[ INFO] [1497251736.595542168]: 3171: this_stamp:1459871506.497979 diff:0.000802
[ INFO] [1497251736.596054259]: 3172: this_stamp:1459871506.498503 diff:0.000524
[ INFO] [1497251736.603382587]: 3173: this_stamp:1459871506.502486 diff:0.003983
[ INFO] [1497251736.603414663]: 3174: this_stamp:1459871506.505802 diff:0.003316
[ INFO] [1497251736.604037222]: 3175: this_stamp:1459871506.506464 diff:0.000662
[ INFO] [1497251736.604695771]: 3176: this_stamp:1459871506.507129 diff:0.000665
[ INFO] [1497251736.612075032]: 3177: this_stamp:1459871506.510447 diff:0.003318
[ INFO] [1497251736.612161457]: 3178: this_stamp:1459871506.514428 diff:0.003981
[ INFO] [1497251736.612629387]: 3179: this_stamp:1459871506.515091 diff:0.000663
[ INFO] [1497251736.613377573]: 3180: this_stamp:1459871506.515755 diff:0.000664
[ INFO] [1497251736.620669252]: 3181: this_stamp:1459871506.517082 diff:0.001327
[ INFO] [1497251736.620714491]: 3182: this_stamp:1459871506.523057 diff:0.005975
[ INFO] [1497251736.621285996]: 3183: this_stamp:1459871506.523720 diff:0.000663
[ INFO] [1497251736.621906861]: 3184: this_stamp:1459871506.524384 diff:0.000664
[... cut ...]
[ INFO] [1497251736.872078155]: 3300: this_stamp:1459871506.774544 diff:0.000638
[ INFO] [1497251736.879408642]: 3301: this_stamp:1459871506.775866 diff:0.001321
[ INFO] [1497251736.879452265]: 3302: this_stamp:1459871506.781840 diff:0.005975
[ INFO] [1497251736.880044689]: 3303: this_stamp:1459871506.782504 diff:0.000663
[ INFO] [1497251736.880706003]: 3304: this_stamp:1459871506.783168 diff:0.000664
[ INFO] [1497251736.888104009]: 3305: this_stamp:1459871506.785158 diff:0.001991
[ INFO] [1497251736.888149669]: 3306: this_stamp:1459871506.790468 diff:0.005309

SirVer, 2017-06-13:
Because our deutsche museum data is YOLO timed - we used an approximation, since we had no hard real-time clock available to us.

sunmk2006, 2017-06-14:
I don't have the concept of touching the knowledge of “YOLO timed",Can you give me some information about it?

SirVer, 2017-06-14:
It is not a scientific term yet, I hope it catches on though: With [Yolo](http://www.urbandictionary.com/define.php?term=Yolo)-timing I mean to say that less thought, care and thoroughness went into time-stamping sensor data compared to extrinsic and intrinsic calibration.

sunmk2006, 2017-06-14:
but if i do not care more on the time-stamping of per-UDP-packet point clouds,when it incorporates [motion estimation](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping_3d/kalman_local_trajectory_builder.cc#L83-L95) that is closely related to the time stamp,Will the result be completely out of control? if not,why?

SirVer, 2017-06-14:
Yes, the result will suffer. For our example bags we have tuned so that the timing jitter is not relevant for the final result, but if you want better results - i.e. smaller voxel size - you require better timing.

sunmk2006, 2017-06-14:
How many parameters can be adjusted to make the final result independent of time jitter? Can you list a few?

SirVer, 2017-06-14:
I do not get the question.

sunmk2006, 2017-06-14:
I use kitti2bag tool to convert a raw data packet (2011_09_26_drive_0002_sync)into a bag file .The output frequency of the velodyne data is 10HZ, The following is the information of each frame，duration = this_stamp - last_stamp，Take second frames as an example ，The time difference between the last frame is about 0.103131s,it contain 12063 data points, Each UDP package is 384 data points，this frame contain 120603 / 384 = 314 udp package, The time per UDP package is 0.103131s / 314 = 328.4μs,but it is 288μs in HDL64E S2 , I can't account for that，can you help me?

[ INFO] [1497249309.084676347]: count:1 duration:0.103127 point_step:16 height:1 row_step:1932832 width:120802 fields_size:4
[ INFO] [1497249309.187220107]: count:2 duration:0.103131 point_step:16 height:1 row_step:1929648 width:120603 fields_size:4
[ INFO] [1497249309.290371504]: count:3 duration:0.103120 point_step:16 height:1 row_step:1932080 width:120755 fields_size:4
[ INFO] [1497249309.393578103]: count:4 duration:0.103140 point_step:16 height:1 row_step:1933648 width:120853 fields_size:4
[ INFO] [1497249309.496675711]: count:5 duration:0.103127 point_step:16 height:1 row_step:1935808 width:120988 fields_size:4
[ INFO] [1497249309.599801937]: count:6 duration:0.103108 point_step:16 height:1 row_step:1937856 width:121116 fields_size:4
[ INFO] [1497249309.703387185]: count:7 duration:0.103115 point_step:16 height:1 row_step:1938544 width:121159 fields_size:4
[ INFO] [1497249309.808613399]: count:8 duration:0.103106 point_step:16 height:1 row_step:1940000 width:121250 fields_size:4
[ INFO] [1497249309.909670002]: count:9 duration:0.103074 point_step:16 height:1 row_step:1939728 width:121233 fields_size:4
[ INFO] [1497249310.012207459]: count:10 duration:0.103063 point_step:16 height:1 row_step:1939920 width:121245 fields_size:4
[ INFO] [1497249310.115291597]: count:11 duration:0.103090 point_step:16 height:1 row_step:1937936 width:121121 fields_size:4
[ INFO] [1497249310.218109733]: count:12 duration:0.103081 point_step:16 height:1 row_step:1936128 width:121008 fields_size:4
[ INFO] [1497249310.321250570]: count:13 duration:0.103073 point_step:16 height:1 row_step:1938256 width:121141 fields_size:4
[ INFO] [1497249310.424477391]: count:14 duration:0.103015 point_step:16 height:1 row_step:1931936 width:120746 fields_size:4
[ INFO] [1497249310.527477149]: count:15 duration:0.103115 point_step:16 height:1 row_step:1929760 width:120610 fields_size:4
[ INFO] [1497249310.630888169]: count:16 duration:0.103229 point_step:16 height:1 row_step:1937904 width:121119 fields_size:4
[ INFO] [1497249310.734187964]: count:17 duration:0.103161 point_step:16 height:1 row_step:1935680 width:120980 fields_size:4
[ INFO] [1497249310.837109915]: count:18 duration:0.103178 point_step:16 height:1 row_step:1937312 width:121082 fields_size:4
[ INFO] [1497249310.940333693]: count:19 duration:0.103202 point_step:16 height:1 row_step:1938672 width:121167 fields_size:4
[ INFO] [1497249311.044198169]: count:20 duration:0.103220 point_step:16 height:1 row_step:1938832 width:121177 fields_size:4
[ INFO] [1497249311.146807449]: count:21 duration:0.103236 point_step:16 height:1 row_step:1939808 width:121238 fields_size:4
[ INFO] [1497249311.249760268]: count:22 duration:0.103239 point_step:16 height:1 row_step:1940352 width:121272 fields_size:4
[ INFO] [1497249311.353170882]: count:23 duration:0.103249 point_step:16 height:1 row_step:1941248 width:121328 fields_size:4
[ INFO] [1497249311.456426435]: count:24 duration:0.103260 point_step:16 height:1 row_step:1942784 width:121424 fields_size:4
[ INFO] [1497249311.559844242]: count:25 duration:0.103244 point_step:16 height:1 row_step:1946800 width:121675 fields_size:4
[ INFO] [1497249311.663060738]: count:26 duration:0.103233 point_step:16 height:1 row_step:1948128 width:121758 fields_size:4
[ INFO] [1497249311.766204208]: count:27 duration:0.103203 point_step:16 height:1 row_step:1950656 width:121916 fields_size:4
[ INFO] [1497249311.869291622]: count:28 duration:0.103198 point_step:16 height:1 row_step:1951616 width:121976 fields_size:4
[ INFO] [1497249311.972616013]: count:29 duration:0.103216 point_step:16 height:1 row_step:1954752 width:122172 fields_size:4
[ INFO] [1497249312.075645450]: count:30 duration:0.103176 point_step:16 height:1 row_step:1954384 width:122149 fields_size:4
[ INFO] [1497249312.178920358]: count:31 duration:0.103170 point_step:16 height:1 row_step:1956960 width:122310 fields_size:4
[ INFO] [1497249312.281975223]: count:32 duration:0.103180 point_step:16 height:1 row_step:1952736 width:122046 fields_size:4
[ INFO] [1497249312.385317480]: count:33 duration:0.103243 point_step:16 height:1 row_step:1954352 width:122147 fields_size:4
[ INFO] [1497249312.488364907]: count:34 duration:0.103213 point_step:16 height:1 row_step:1956288 width:122268 fields_size:4
[ INFO] [1497249312.591913025]: count:35 duration:0.103146 point_step:16 height:1 row_step:1956320 width:122270 fields_size:4
[ INFO] [1497249312.694748915]: count:36 duration:0.103223 point_step:16 height:1 row_step:1957088 width:122318 fields_size:4
[ INFO] [1497249312.798161319]: count:37 duration:0.103275 point_step:16 height:1 row_step:1956912 width:122307 fields_size:4
[ INFO] [1497249312.901340552]: count:38 duration:0.103279 point_step:16 height:1 row_step:1957424 width:122339 fields_size:4
[ INFO] [1497249313.004436701]: count:39 duration:0.103252 point_step:16 height:1 row_step:1959584 width:122474 fields_size:4
[... cut ...]
[ INFO] [1497249316.409768522]: count:72 duration:0.103119 point_step:16 height:1 row_step:1957680 width:122355 fields_size:4
[ INFO] [1497249316.513556770]: count:73 duration:0.103114 point_step:16 height:1 row_step:1954336 width:122146 fields_size:4
[ INFO] [1497249316.616466326]: count:74 duration:0.103108 point_step:16 height:1 row_step:1952656 width:122041 fields_size:4
[ INFO] [1497249316.719080363]: count:75 duration:0.103093 point_step:16 height:1 row_step:1955344 width:122209 fields_size:4

wohe, 2017-06-16:
I'm not sure I got your last question. Could you explain what you observe and what you would have expected instead?

Keep in mind though that the timing of the laser firings is pretty accurate, so you would expect each UDP packet to contain information about a certain fixed duration. The 10 Hz rotation speed is something that the LIDAR tries to maintain, but is a lot less accurate. Rotation speed varies during each revolution, and different revolutions can take more or less time.

sunmk2006, 2017-06-17:
The question I really want to ask is how to properly divide the laser data frames（which is 10 frames per second in kitti datasets） into UDP packets.From the data sheet of velodyn, i know  each  udp package should contain 384 points and  the firing time of each udp packet will take  288μs in HDL64E S2, However, these two standards can not be correctly divided in the kitti bag file, Can you help me?

SirVer, 2017-06-19:
Just cut the data arbitrarily in smaller pieces? You do not have to hit the UDP packet size exactly.

sunmk2006, 2017-06-19:
In this case, the slice time also needs to be equally distributed，no longer limited to the firing capability of the equipment，yes?

SirVer, 2017-06-19:
Yes. It is impossible to get to the original firing pattern from the data kitti released IMHO. If you want that, you need to reach out to andreas and ask for the original UDP packets (which would of course be very useful to others as well).

sunmk2006, 2017-06-19:
ok,thank you very much!

wzw19940427, 2019-08-09:
> hi,do you know [the KITTI odometry dataset](http://www.cvlibs.net/datasets/kitti/eval_odometry.php) , i want to use it to do 3D cartographer accuracy tests , A few problems come up ：
>
>     1. the [...]

Hi, sorry for disturbing. I try to use the kitti odometry datasets to run cartographer,but i can't get a proper result,not to mention comparing the output result with the ground truth.I try to use some tools like kittiplayer and kittihelper,but the result in rviz is wrong clearly,I don't know where the problem is,if you have got the proper output result,would you please give me a reference about it? 
Thanks a lot!