cartographer-project/cartographer_ros #343: Support multi trajectory

SirVer, 2017-05-17:
googlecartographer/cartographer_ros/#335 started making the node multi trajectory aware. This leaks some problematic abstractions:

1) When a trajectory is finished, the node writes some derived assets (Trajectory + OccupancyGrid or X-Rays + PLY) and starts a new one. The assets use `trajectory_builder_options` for laser insertion options and resolution (which one should this be for multi trajectory?). The trajectory proto only contains the current opinion of one trajectory.
2) BuildOccupancyGrid2D - which is used to publish a legacy OccupancyGrid for ROS.

I think we have to rethink what we want to do here. I think writing the assets is fundamentally wrong in the node - instead, it should only dump the sparse pose graph (with or without submaps, as desired) and all further asset generation should be deferred to assets_writer_main. This would allow us to hold much less data in memory while slamming. 

This would also mean to pull OccupancyGrid generation into a separate ROS node that translates between Cartographer's representation and legacy ROS Occupancy grids. I am unclear if this node should keep laser data around or just reuse the submaps from cartographer and draw them on-top of each other (similar to what the visualization currently does).

/cc @macmason @wohe @yutakaoka Thoughts?

wohe, 2017-05-18:
Sounds good. We might also want to separate finishing a trajectory from writing the current SLAM state. Maybe we should move this to a separate service?

yutakaoka, 2017-05-23:
For now(https://github.com/googlecartographer/cartographer_ros/pull/346), I will create `write_assets` service to separate it from `finish_trajectory` service.

commit 1ea5f90b (2017-07-21): New node that creates a ROS map from submap textures. (#432)

Fixes #343.
closed 2017-07-21 by SirVer