cartographer-project/cartographer_ros #350: scans_per_accumulation

sunmk2006, 2017-05-22:
TRAJECTORY_BUILDER_3D.scans_per_accumulation = 160
If I keep decreasing this value, the trajectory will become more and more dense， Will the amount of computation increase?

SirVer, 2017-05-26:
This purely depends on your setup, the answer is likely yes. But this is not really a tuning parameter, usually there is a single value or small range that is correct for your setup.

See this [FAQ](https://google-cartographer-ros.readthedocs.io/en/latest/faq.html#the-laser-data-in-the-3d-bags-is-much-higher-than-the-maximum-reported-20-hz-rotation-speed-that-the-vlp-16-can-do-why) entry.

closed 2017-05-26 by SirVer

sunmk2006, 2017-05-27:
_which incorporates motion estimation by combining constant velocity and IMU measurements, for matching.Since there are two VLP-16s, 160 UDP packets is enough for roughly 2 revolutions, one per VLP-16_
Does it mean that lower or higher TRAJECTORY_BUILDER_3D.scans_per_accumulation affects trajectory accuracy?

SirVer, 2017-05-29:
Yes.