cartographer-project/cartographer_ros #351: Configuration and loop closure problems

HolmarSeffelgruber, 2017-05-24:
Hi! I am trying to use cartographer for high definition pointcloud maps with 1-2 VLP16 and a Microstrain 3DM GX3 25 IMU. We want to use the produced maps in mcl localization using the same setup. We want the localization as accurate as possible. I got trouble with rotated maps and loop closure and the urdf setup, although the loop closure seems to work better with the search window from [ozgudum](https://github.com/SirVer/cartographer_ros/commit/09c8aa6746034aba9b0a06662b3889d336183422).

In these bagfiles the Velodynes are rotating by 1200rpm. I noticed it works better with higher rpm when I used different rpm in the front and back and compared the single sensor maps.
1. You are using a high number of `TRAJECTORY_BUILDER_3D.scans_per_accumulation`. I get it to work best using 1-10 accumulations. Do you have an idea what the reason might be or is it not important?

2. I get good results by adjusting the parameter `TRAJECTORY_BUILDER_3D.ceres_scan_matcher.rotation_weight` but I didn't see someone else using this parameter. I thought maybe there is something else wrong with my setup and you might help out.

3. I found the parameter imu_gravity_time_constant. Since I am moving with a car I think I should possibly increase this constant since the acceleration is higher compared to a robot or walking.

4. In the pointcloud maps I create using only one lidar I get more details when I look at the cars. The reason is probably a disalignment in the transformations. Can you give me a recommendation how I can calibrate it to the physical joints or rather their representation in urdf. I think my measurements are about as accurate as I can get them. This might be the biggest problem I guess

5. Are the high-/ and low resolution parameters important for the resolution/accuracy of the ply pointcloud map?
Bag-Files: ges: I'm driving up a garage but loop closure will probably not happen since I don't visit a spot twice. I used it to calibrate rotation weight.
loop_closure_test: same garage but I switch between the floor a few times. I was hoping to get some good loop closure examples here.
parking_area: the loop closure works pretty good here only using the front Lidar but not using both Lidar.

I'm sorry I cant provide a Github rep but these files should work with [#342](https://github.com/googlecartographer/cartographer_ros/pull/342)
[Bagfiles and configuration](https://fh-aachen.sciebo.de/index.php/s/h9s1cLQKSNhDz52)

SirVer, 2017-05-26:
> We want to use the produced maps in mcl localization using the same setup.

We cannot help you there. No experience with this. We are working on pure localization using only Cartographer right now though.

About RPM: higher is better imho. I never saw a reason for slower rotation.

Answers to your individual questions:

1. See [this FAQ entry](https://google-cartographer-ros.readthedocs.io/en/latest/faq.html#the-laser-data-in-the-3d-bags-is-much-higher-than-the-maximum-reported-20-hz-rotation-speed-that-the-vlp-16-can-do-why). You are likely loosing quality by not giving the SLAM a chance to unwarp your laser revolutions.

2. It means that the scan matcher has to pay more for rotating which can be a perfectly valid thing to tune for.

3. Ah... what a terrible parameter, I wish it wasn't there. We never change that, also not on cars. I have not much experience changing it.

4. Cartographer team never properly investigated calibrating two distinct devices with respect to each other. 

I am pretty convinced that you cannot do a good enough job by just measuring them or relying on CAD. An error as small as a degree can be too much error. This also immediately leads to the question if your setup is stiff enough - maybe your lasers shake too much?

We think calibration could probably be done in the optimization problem, but an offline step is preferable. A valid and simple approach could be to SLAM both lasers together with a YOLO extrinsic, use the trajectory to generate small point clouds (so that they still fit into memory), but large enough that they do have overlap and then try to IPC the clouds with respect to each other. This should give an approximation of your calibration and the results should be better when your system is shaking due to better random sampling. Requirement is some overlap in your field of views of the lasers.

5. Yes, totally. But the relationship is not straightforward or even mathematically expressable. The high resolution defines the resolution of the finest grid Cartographer is using. Let's say this is 5cm - this means the correlating scan matcher (for loop closure) will find matches up to 5cm of accuracy. The ceres scan matcher is used with this as a starting point and it works over a cubic interpolation of this 5cm grid. This means that as long as your interpolation represents reality well, you should get very similar results in the quality of the final trajectory - but having 5cm instead of 15cm means you have more data for the interpolation, ergo (probably) a more correct approximation. 

I now looked at your data and tuned a bit. You can find my tuning in the [last few commits in this branch](https://github.com/SirVer/cartographer_ros/commits/user/HolmarSeffelgruber). It came down to lowering the loop closure score because to few constraints where found. And for the two vlp16 case you accumulated 10 revolutions (which comes down to 5 for each, since your data is so nicely timed) instead of 2. That insta fixed that. I only looked at the `parking_area` bag due to time constraints, but I think the settings should be fine.

Below is a 8MB large png that was generated with the asset writer and shows one lidar in green and the other in red. You can see that your calibration is pretty decent. You can also see this by investigating the generated point cloud (created by writing a ply and using [googlecartographer/point_cloud_viewer](https://github.com/googlecartographer/point_cloud_viewer)). Some impressions are also attached below.

Let me know if you have further questions.

[image]

[image]
[image]
[image]

closed 2017-05-26 by SirVer

HolmarSeffelgruber, 2017-05-29:
Thank you for your reply it looks very promising!
I didn't want to ask for your help on localization. I was just giving some information about my case.

1. What do you mean by unwarping the lasers?

5. I was using the pcl library up until now to visualize my pointclouds but your way looks to give more details using the full potential of the pucks. 
However I have a few problems:

A. The Assets writer crashes at some point even though it creates a ply and the xrays so it may be work normally. I guess you used a cartographer version after you added the assetswriter 3d? These are the last messages:
`I0529 10:14:19.769731 10802 assets_writer_main.cc:179] Processing pipeline...
I0529 10:14:19.847920 10802 assets_writer_main.cc:203] Processed 0 of 369.237 bag time seconds...
I0529 10:17:33.015483 10802 counting_points_processor.cc:43] Processed 164601187 and finishing.`

B. I don't know how to use the pointcloud viewer. I haven't used rust yet. I succeded in installing all the programs and built the binaries. I installed the most recent stable nvm.

SirVer, 2017-05-29:
> I didn't want to ask for your help on localization. I was just giving some information about my case.

ok. I wanted to convey that you might not need amcl in the future anymore, once Cartographer can do pure localization.

1. I mean fixing the *moving shutter* effect of the rotating laser: your car moves while your laser rotates. If you pass a full revolution to the SLAM, it cannot account for that. If you pass it lots of smaller point clouds and tell it to accumulate them internally to a bigger one, it can *unwarp* the rotation by assuming a linear motion extrapolated from the last SLAM state.

A. It did not crash for me, though we probably used similar command lines. Could you try investigating where the crash occurs? My hunch is that this is the XRayPointsProcessor::cleanup method.

B. After running `cargo build --release`, the binaries are in `target/release`. You should be fine by using `build_octree file.ply --output_directory octree` and then running `viewer octree` after. Then open chrome (other browsers are not tested) to `localhost:5433`. You will see initially black, just start moving using WASD and click and drag to rotate. Mouse wheel adjusts speed.

HolmarSeffelgruber, 2017-05-29:
I tried to make fewer xrays and didn't create the ply another time but it doesn't change the shutdown. But as I get the ply, a grey and a colored xray I wonder If it's not working fine.
I also can't see a node for the assets writer which makes it difficult using the rqt_console for debugging since I can't set the debugging level for the assets_writer.

In the text file you can see that the last command before assets_writer dies is spin. Do you have an idea for debugging?

[image]
[roslaunch-a234-12591.txt](https://github.com/googlecartographer/cartographer_ros/files/1035977/roslaunch-a234-12591.txt)

SirVer, 2017-05-30:
Actually, the text file says the `process has finished cleanly`. Since it is just supposed to run through the bag once and terminates, it probably is working as intended. Sorry for the misunderstanding.

HolmarSeffelgruber, 2017-06-12:
Seems like the last week was very productive for you. There are some nice new features. I'm especially happy about the trajectory visualization.
There is a flaw for me though. The configuration you tuned doesn't work as well as before anymore. I think the changes had a big effect on the SLAM as well. 
Is it the same for your demos or is it just my setup?

SirVer, 2017-06-12:
My gut feeling is that this is probably due to googlecartographer/cartographer#307, googlecartographer/cartographer#308 or googlecartographer/cartographer#309. They probably will require a bit of retuning, but are in general a good step towards simplification of the tuning process. Sorry for the inconvenience.

HolmarSeffelgruber, 2017-06-12:
Your gut hat the correct feeling. I used the old values for `SPARSE_POSE_GRAPH.optimization_problem.acceleration_weight = 1e4
SPARSE_POSE_GRAPH.optimization_problem.rotation_weight = 3e6` 
and nearly got my good scans back. This might help some others, too.
I will play with some more parameters. Thank you for your work!

OneYouth, 2017-09-07:
I met the same problem as follow when I run    
"roslaunch cartographer_ros assets_writer_backpack_3d.launch bag_filenames:=/home/hao/Downloads/b3-2016-04-05-15-51-36.bag pose_graph_filename:=/home/hao/Downloads/b3-2016-04-05-15-51-36.bag.pbstream"
[image]
Could you please tell me how to solve it ? Thanks.

ikhann, 2020-06-17:
> Hi! I am trying to use cartographer for high definition pointcloud maps with 1-2 VLP16 and a Microstrain 3DM GX3 25 IMU. We want to use the produced maps in mcl localization using the same setup. We [...]

Hi,
Could you please load a mentioned bag file here again? I need it for testing my tune process.