cartographer-project/cartographer_ros #353: constraint_builder

sunmk2006, 2017-05-26:
In the latest version of cartographer code, the variable constraint_transform is defined:
[transform::Rigid3d constraint_transform](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping_3d/sparse_pose_graph/constraint_builder.cc#L220)
but I used an earlier version,it is defined:
  ceres::Solver::Summary unused_summary;
  ceres_scan_matcher_.Match(
      pose_estimate, pose_estimate,
      {{&filtered_point_cloud, submap_scan_matcher->hybrid_grid}},
      &pose_estimate, &unused_summary);

  const transform::Rigid3d constraint_transform =
      submap->local_pose().inverse() * pose_estimate;
 Why is there such a change?

SirVer, 2017-05-26:
I am not sure which change you are referring to specifically - links to source code would help.

But my guess is that the change you see comes from a recent simplification: we are operating as much as possible in the submap frame while before we were often operating in the odometry (or `local`) SLAM frame.

closed 2017-05-26 by SirVer

sunmk2006, 2017-05-26:
[latest_version](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping_3d/sparse_pose_graph/constraint_builder.cc#L219)
ceres::Solver::Summary unused_summary;
  transform::Rigid3d constraint_transform;
  ceres_scan_matcher_.Match(
      pose_estimate, pose_estimate,
      {{&filtered_point_cloud, submap_scan_matcher->hybrid_grid}},
      &constraint_transform, &unused_summary);

  constraint->reset(new OptimizationProblem::Constraint{
      submap_id,
      node_id,
      {constraint_transform,
       1. / std::sqrt(options_.lower_covariance_eigenvalue_bound()) *
           kalman_filter::PoseCovariance::Identity()},
OptimizationProblem::Constraint::INTER_SUBMAP});

my used version:
  ceres::Solver::Summary unused_summary;
  ceres_scan_matcher_.Match(
      pose_estimate, pose_estimate,
      {{&filtered_point_cloud, submap_scan_matcher->hybrid_grid}},
      &pose_estimate, &unused_summary);

  const transform::Rigid3d constraint_transform =
      submap->local_pose().inverse() * pose_estimate;
  constraint->reset(new OptimizationProblem::Constraint{
      submap_index,
      scan_index,
      {constraint_transform,
       1. / std::sqrt(options_.lower_covariance_eigenvalue_bound()) *
           kalman_filter::PoseCovariance::Identity()},
      OptimizationProblem::Constraint::INTER_SUBMAP});

SirVer, 2017-05-26:
I already gave the answer: before we did the scan matching in the local pose (odom frame), now we do it in the submap frame.

sunmk2006, 2017-05-27:
Sorry, I haven't fully understood the various transform in cartographer.