cartographer-project/cartographer_ros #354: missing_data_ray_length has no effect?

romainreignier, 2017-05-29:
Following the discussion found in the Google group [here](https://groups.google.com/forum/#!topic/google-cartographer/uNL7WM2XXnc), I have setup  a simple environment in the Gazebo simulator to test the effect of the parameter `missing_data_ray_length` of the `TRAJECTORY_BUILDER_2D`.

In the simulated environment, there is a wall on the left of the robot and nothing on the right. While moving along the wall, only the left side of the map becomes white (Free space) and nothing on the right.

The following settings are used:

```lua
TRAJECTORY_BUILDER_2D.min_range = 0.05
TRAJECTORY_BUILDER_2D.max_range = 10.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 8.
```

In the simulator, when no object is hit within the sensor range, a `inf.` value is returned, which is greater than the sensor max range. So, according to the documentation [here](https://github.com/googlecartographer/cartographer/blob/master/docs/source/configuration.rst#cartographermapping_2dprotolocaltrajectorybuilderoptions):

> float missing_data_ray_length
>    Points beyond 'max_range' will be inserted with this length as empty space.

The space between the robot and the distance set by the `missing_data_ray_length` should become white (Free space).

The result in RViz:
[image]

The workspace used and a roslaunch ([`gazebo.launch`](https://github.com/romainreignier/cartographer_ros/blob/missing_data_ray_length/cartographer_ros/launch/gazebo.launch)) can be found in [my fork](https://github.com/romainreignier/cartographer_ros/tree/missing_data_ray_length), and a rosbag is uploaded [here](https://drive.google.com/file/d/0B37uoIn2q4qqVUZlQUlldWwzNVk/view?usp=sharing).

From my understanding of the source code, because of [this line](https://github.com/googlecartographer/cartographer_ros/blob/cd1276a99ccffcc491d0b2e50296ec04347ba5f2/cartographer_ros/cartographer_ros/msg_conversion.cc#L110), any value that is not in the range between `range_min` and `range_max` is dropped so the miss information is lost.

SirVer, 2017-05-30:
The [LaserFan documentation](http://docs.ros.org/api/sensor_msgs/html/msg/LaserScan.html) states:

    range data [m] (Note: values < range_min or > range_max should be discarded)

So I think our code is actually doing the right thing.

The use of the `missing_data_ray_length` is in the SLAM is in [TrajectoryBuilder2D](https://github.com/googlecartographer/cartographer/blob/4bb02c724057c8eae618d5c7b8d039c5dfb22264/cartographer/mapping_2d/local_trajectory_builder.cc#L41). To see the effect of this parameter, you have to get gazebo to return a value in the valid range of what it claims the sensor returns, i.e. < `inf`, but larger than `TRAJECTORY_BUILDER_2D.max_range` and then you should see free space inserted till `TRAJECTORY_BUILDER_2D.missing_data_ray_length`. 

Closing, as I do not see a bug here. Please let me know if you disagree and we can reopen.

romainreignier, 2017-05-30:
I have added [a simple node](https://github.com/romainreignier/cartographer_ros/blob/53718c7826eb60f1e3c8ec3c6fdfa057ab87e91b/cartographer_ros/src/laser_filter.cc) that reduce the miss fans to `range_max - 0.01` and now it works. Thanks.

[image]

So it is not very intuitive and will never work with a default laser driver (a miss is either `inf.` either `range_max` or `range_max + 1.0` in most drivers) but I understand the choice to respect the `LaserScan` documentation.

I hope this issue will help other people struggling like me on this behavior for a long time.

SirVer, 2017-05-31:
> So it is not very intuitive and will never work with a default laser driver...

I find this is an unfair statement and I challenge that our design is not intuitive. I agree that ROS design is not, and it is under-documented, but I argue below that it allows for encoding all state. I think NaN and INF are not intuitive - and ROS does not document what they should mean. Some drivers for example report INF for bogus return (object too close).

Lets recap what a laser can actually report:

1) No return -> too long a signal.
2) Bogus return -> probably something to close or overexposure through another device.
3) valid return 

ROS does not specify how to encode this specifically. It does only state though that everything outside `min_range` and `max_range` should be discarded. This allows to encode all of these statements quite intuitively:

1) `max_range`
2) `min_range`
3) `min_range` <= val <= `max_range`

That said, Cartographer (the library) does not deal with this convention, only cartographer_ros has to. Internally, we only deal with point clouds. It is a better representation, because it implies only valid readings and implies angles. The `min_range` and `max_range` you are setting are a Cartographer construct - not bound by the definitions of ROS.

closed 2017-05-31 by SirVer

romainreignier, 2017-06-01:
> I find this is an unfair statement and I challenge the statement that our design is not intuitive. I agree that ROS design is not, and it is under-documented

Yes, I am not arguing against Cartographer design that is great. I agree with you that the ROS design is not so clear.

About your segmentation, I think the **valid return** should be

```c
min_range <= val < max_range
```

Without the `=` for `range_max` and then use `range_max` for **no return**. But I agree that it may cause issues because it is not self-explanatory.