cartographer-project/cartographer_ros #355: Zesty/ROS Lunar build

ojura, 2017-05-30:
Hi, I was trying to build Cartographer on Ubuntu 17.04/ROS Lunar.

I stumbled upon a linking error related to a dependency of the PCL library, the MPI library. I believe this is a bug in the `libpcl-dev` package on Zesty. I described the error, including a minimal working example, on the PCL-users mailing list:
http://www.pcl-users.org/Build-failure-on-Ubuntu-17-04-td4044552.html

I also found a really dirty fix, which is to modify the system's PCL CMake module config file to remove the offending library (described in the reply to myself on the mailing list). 

After this, I managed to build Cartographer without problems :-) Maybe you guys could chime in, I'm wondering where should I report this.

SirVer, 2017-05-30:
Is there no way of installing said dependency somehow properly?

ojura, 2017-05-30:
Everything is already installed when you have installed ROS. The issue is not that it's missing, but rather that CMake sees two libmpi.so files (one in /usr/lib/x86_64-linux-gnu/openmpi/lib, one in /usr/lib) and refuses to create a rule for linking against that library. This in turn causes make to fail because of the missing rule. When you remove the /usr/lib version of libmpi from the PCL CMake module config file as I described (it wasn't there in Xenial), there are no more issues with linking PCL.

I haven't verified if this happens on Debian though, but I will keep in mind to contact the Debian maintainers if the folks at Launchpad don't respond.

Since this is not a Cartographer issue, this issue can be closed, or alternatively it can be kept open until this issue with PCL is resolved.

SirVer, 2017-05-30:
I'll close it since this is not actionable for us. It seems like we cannot even work around it. It is good to know in case somebody else runs into this problem down the line. Thanks for reporting.

closed 2017-05-30 by SirVer

ojura, 2017-06-23:
@SirVer since this week has passed for me mostly fixing Zesty build issues (a very fun and satisfying experience - not :) ), let me return to this one, the last remaining bug which prevents a proper build on Zesty.

> not actionable for us

I have just found that it actually is actionable. This is how they mitigated the problem in ros-perception, by doing the same thing I proposed above (removing libmpi.so from VTK_LIBRARIES), just in CMake:
https://github.com/ros-perception/perception_pcl/commit/42683bd4069c3a8bcec56d6c4e686d9e8643e77b

I've tried reaching out to PCL folks but got no response, which was very disappointing. I am not sure if you would want to include something like this, as it is really an ugly CMake hack - but on the other hand, if ros-perception folks did it, maybe we could too.

SirVer, 2017-06-23:
Yes, let's do it. With a proper comment and a reference this should be fine. Should PCL ever fix this, we can remove the workaround again.

ojura, 2017-06-23:
Sure. The reference URL (https://github.com/ros-perception/perception_pcl/blob/b53b5d5c3abc7c3afde8a77177674e6d74c59522/pcl_ros/CMakeLists.txt#L12) will be quite long, do you want me to use an URL shortening service?

SirVer, 2017-06-23:
No, just paste it without line breaks.

ojura, 2017-06-23:
Okay, done.

PR #396 (2017-06-26): Fix PCL linking issue in Zesty
Fixes #355.
commit 125aee30 (2017-06-26): Fix PCL linking issue in Zesty (#396)

Fixes #355.