cartographer-project/cartographer_ros #356: Strange tf behaviour

Panda-cool, 2017-05-30:
Hello,

I'm working on map generation with different tools and I currently struggle with google cartographer for ROS. Only the first scan of the map appears on rviz and when I check my TF frames, the robot frames (laser, base_link, etc) stay at the same position while my odom frame is "moving" backward w.r.t. the robot movement in the simulator. It actually works fine with gmapping so I don't know what goes wrong here.

Here are my launch and config file for cartographer https://drive.google.com/open?id=0B1c... https://drive.google.com/open?id=0B1c...

And here is my TF tree if that can help but it seems all right https://drive.google.com/open?id=0B1c...

and finally here is a small bag file https://drive.google.com/open?id=0B1cH0l8kxYjZb1FOUS1kbDZoZHc

I think there is a problem with tf somewhere because roswtf show me this but I can't figure out where: 
WARNING The following node subscriptions are unconnected: 
* /cartographer_node: 
     * /tf_static 
* /rviz_1496128181484858929: 
    * /tf_static 
    * /map_updates

I'm relatively new with ROS, sorry if I misunderstand something.

Thank you in advance for your help

SirVer, 2017-05-30:
@macmason Could you have a look?

macmason, 2017-06-01:
@SirVer happy to look, but the earliest I'll have a chance is next Tuesday.

Panda-cool, 2017-06-06:
@macmason Hello, did you find what goes wrong ?

macmason, 2017-06-06:
@Panda-cool hey, just got a chance to take a look. Several of the links in your original report are broken...it looks like they got truncated somehow.

Panda-cool, 2017-06-06:
This is indeed strange as it works perfectly fine with gmapping. Maybe I miss something in the cartographer launch file but I couldn't say what.

macmason, 2017-06-06:
...huh?

I can't download your files, because the links are broken. Without files, I can't debug further.

Panda-cool, 2017-06-06:
Oh sorry it's quite late here, I though you were talking about the roswtf report. Those links should work https://drive.google.com/open?id=0B1cH0l8kxYjZdXRITmdpZmFaLTQ
https://drive.google.com/open?id=0B1cH0l8kxYjZU1ZqNFJYRV9FT28
https://drive.google.com/open?id=0B1cH0l8kxYjZeXliWkphOTE2RlU
https://drive.google.com/open?id=0B1cH0l8kxYjZb1FOUS1kbDZoZHc

Panda-cool, 2017-06-06:
In the case the links are still broken, tell me, I will find some other media than drive.

macmason, 2017-06-06:
So I've eyeballed this a bit, and here's my first guess:

Your bag file contains transforms between `odom` and `map`. In your tf graph, `cartographer_node` is _also_ providing transforms between `odom` and `map`. That means that when I play back your bag with cartographer running, they compete, so the transform's inconsistent. 

(If the `map` <-> `odom` transform _in the bag_ is already from Cartographer, then you have something else wrong...but the technique described below is still how you should start.)

A similar problem occurs in the PR2 example dataset; look at `demo_pr2.launch` to see what to do about it.

Panda-cool, 2017-06-06:
Actually the transform comes from cartographer, I checked to be sure and I have no transform between map and odom without launching it.
However I checked the demo_pr2.launch file, you meant I should use the tf_remove thing ? (I tried, did not work unfortunately)
Also, when I use gmapping, I have to set the minScore parameter to 50 to avoid a behaviour similar to the one I experience with cartographer (because it seems I have some bad scans in between the good and that makes the robot jump). Could there be a link ? I did not find a parameter similar to the gmapping minScore to test it, is there one ?

macmason, 2017-06-06:
> Actually the transform comes from cartographer, I checked to be sure and I have no transform between map and odom without launching it.

There is a map / odom transform in your bag file:

```
---
transforms: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1496145366
        nsecs:  73816500
      frame_id: map
    child_frame_id: odom
    transform: 
      translation: 
        x: -0.0110359287661
        y: -0.000684995016783
        z: 0.0
      rotation: 
        x: 0.0
        y: 0.0
        z: -0.000644535176762
        w: 0.999999792287
---
```

(from `rostopic echo tf`)

So I'm not sure what you mean.

Panda-cool, 2017-06-06:
I mean I stupidly recorded the bag while cartographer was running, when it is not running, there is no such transform in the tf tree. Do you want me to provide you another bag without that ?

Panda-cool, 2017-06-06:
(I need some sleep so here it is in cas you need it https://drive.google.com/open?id=0B1cH0l8kxYjZMXpJc2ZfRWE2dG8)

SirVer, 2017-06-07:
You can use [this script snippet](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/launch/demo_pr2.launch#L29) in your launch file to remove the TF messages while playing back. The script ships with Cartographer.

Panda-cool, 2017-06-07:
Yes, as suggested by @macmason I tried to add the tf_remove_frames node to my launch file, unfortunately without any improvement.

SirVer, 2017-06-07:
Oh sorry, I did not see that suggestion from Mac. On point, as always. Sorry for that noise.

macmason, 2017-06-07:
In the newly-posted bag, something's strange about your laser scan. Messages arrive, but the actual shape of the scan never changes (which is going to make Cartographer fail). That shouldn't work in GMapping, either.

Something very similar is happening in your original bag as well; rolling robot, fixed laser scan.

Panda-cool, 2017-06-07:
I recorded the bag in a simulator where the robot moves forward in a straight corridor so I think it's normal the data doesn't change much. However, when I was working with gmapping I more or less understood that some scans were "bad" and were decreasing the average matching. So as I said above, I set some parameter in order to drop those scans and it works fine now. 
But if cartographer needs the data to change should I make the robot moves in zigzag or something like that ?

Panda-cool, 2017-06-07:
Maybe there is something I'm missing about the laser scans though...

Panda-cool, 2017-06-07:
Maybe I'm missing something about the laser scan, here is again a new bag when I made the robot move approximately in zigzag (and it works with gmapping while in Cartographer I'm stuck in the first position, my odom moving backward, etc)
https://drive.google.com/open?id=0B1cH0l8kxYjZb2xDVEV6Rm1IQ00

macmason, 2017-06-07:
Ok, next question: what sort of laser is this? The returns are 10 degrees apart, which is extremely sparse.

Panda-cool, 2017-06-07:
There is actually no laser on the robot so it is emulated using the 8 proximity sensors : the values are recomputed in order to have 19 virtual sensors indeed separated by 10 degrees. I can compute more or them if needed, what do you think is good for Cartographer ?

macmason, 2017-06-07:
My instinct is that that will not be enough, particularly since you really only have eight readings (interpolating between them isn't going to produce new data, after all).

The next thing to try would be a much lower resolution, with bigger uncertainties, but I'm really in doubt that this is going to work.

For GMapping: have you tried longer runs? My guess is that it won't be able to handle anything large either.

Panda-cool, 2017-06-08:
Ok I will try that, could you remind me how to set resolution and uncertainty parameters in Cartographer ?

For GMapping, The biggest run I tried is as 2x2 meters square corridor and it works. I don't think I will try mapping something bigger than 5x5 with such a small robot, but then maybe Cartographer won't work well is that it ?

macmason, 2017-06-08:
See the contents of `trajectory_builder_2d.lua`, in particular the `submaps.resolution` field. 

However, this problem really isn't well-posed: you don't have enough data, and your environment is tiny. That GMapping works is just good luck, but nobody's ever really tried to make Cartographer handle this kind of case, and it's not going to be a priority.

macmason, 2017-06-08:
@SirVer I don't seem to have closing powers here.

SirVer, 2017-06-09:
@macmason Should be fixed. Policy only allows me to add individuals though, so the rest of TRI must be added one by one should that be desired.

closed 2017-06-09 by SirVer