cartographer-project/cartographer_ros #357: no map generated using kinect2

CTTC, 2017-05-31:
I want to run cartographer with kinect2. Here are my configuration file and launch file.
```bash
<launch>
  <param name="/use_sim_time" value="true" />
  <node name="squash_depth_camera" pkg="depthimage_to_laserscan"
      type="depthimage_to_laserscan">
    <param name="output_frame_id" value="kinect2_ir_optical_frame"/>
    <remap from="image" to="/kinect2/qhd/image_depth_rect" />
    <remap from="camera_info" to="/kinect2/qhd/camera_info" />
  </node>
<!--   <node name="squash_depth_camera" pkg="pointcloud_to_laserscan"
      type="pointcloud_to_laserscan_node">
    <remap from="cloud_in" to="/kinect2/qhd/points" />
  </node> -->
  <node name="cartographer" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename kinect2.lua"
      output="screen">
  </node>
  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros
          )/configuration_files/kinect2.rviz" />
  <node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)" />
</launch>
```

```bash
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "kinect2_ir_optical_frame",
  published_frame = "kinect2_ir_optical_frame",
  odom_frame = "odom",
  provide_odom_frame = false,
  use_odometry = true,
  use_laser_scan = true,
  use_multi_echo_laser_scan = false,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true
MAP_BUILDER.num_background_threads = 7

TRAJECTORY_BUILDER_2D.min_range = 0.1
TRAJECTORY_BUILDER_2D.max_range = 4.0
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 2.0
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.motion_filter.max_time_seconds = 0.3
TRAJECTORY_BUILDER_2D.motion_filter.max_distance_meters = 0.2
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(3.)
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight = 70
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight = 300

TRAJECTORY_BUILDER_2D.submaps.resolution = 0.035
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 120
SPARSE_POSE_GRAPH.optimize_every_n_scans = 120
SPARSE_POSE_GRAPH.constraint_builder.min_score = 0.82
SPARSE_POSE_GRAPH.constraint_builder.sampling_ratio = 1.
SPARSE_POSE_GRAPH.constraint_builder.ceres_scan_matcher.covariance_scale = 3e-5

return options
```

I have uploaded these two files in my forked directory: `https://github.com/CTTC/cartographer_ros`
And a small bag recording data published by kinect2 is available [here](https://my.pcloud.com/publink/show?code=XZ6xj3ZugtWj3XSLXpcAKgXAR2Cyua3ny37)

CTTC, 2017-06-01:
Thanks, @Entropy512  I misunderstood the meaning of `provide_odom_frame`.Now I have set use_odometry to false as well. And I can get map message now, but it's a pretty bad map. As `imu` and `odom` are optional in 2D case, I am only using a kinect2 now.  So I only have these frames available: kinect2_ir_optical_frame, kinect2_rgb_optical_frame,kinect2_link. No other frames are available.  In this case, how should I set the `map_frame`, `tracking_frame`, and `published_frame`? If I use the lua file shown above, the `/scan` data shown in rivz is always blinking. It shows `Transforms Ok` and `For frame [kinect2_ir_optical_frame]: No transform to fixed frame [map].  TF error: [Could not find a connection between 'map' and 'kinect2_ir_optical_frame' because they are not part of the same tree.Tf has two or more unconnected trees.]` alternately.

SirVer, 2017-06-01:
You are doing 2D SLAM with kinect? Are you projecting the points into a plane before passing it to Cartographer somehow?

`map_frame` should be "map", `tracking_frame` should be the frame which's position you are interested in. Set `published_frame` to the same thing.

You do not need to mention `/map` in your TF tree, but it seems something else is misconfigured. Did you run `roswft`?

CTTC, 2017-06-01:
@SirVer  Thanks for your reply. Yes, I am trying 2D SLAM with kinect (no other sensors for now). I followed the turtlebot example which convert the depth image into laser scan using `depthimage_to_laserscan`.

I can run the program now. It's just that the map does not look good (not even close). I assume that is because no imu or odom data is available. 

Also, I tried your advice, and here is the output of `roswtf`:
```bash
Found 2 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /squash_depth_camera:
   * /clock
 * /cartographer:
   * /tf_static
   * /clock
 * /rviz:
   * /tf_static
   * /clock
   * /map_updates

WARNING Received out-of-date/future transforms:
 * receiving transform from [/cartographer] that differed from ROS time by 1496307025.67s
 * receiving transform from [/kinect2] that differed from ROS time by 1496307025.71s

Found 3 error(s).

ERROR /use_simtime is set but no publisher of /clock is present
ERROR TF re-parenting contention:
 * reparenting of [kinect2_ir_optical_frame] to [kinect2_rgb_optical_frame] by [/kinect2]
 * reparenting of [kinect2_ir_optical_frame] to [map] by [/cartographer]

ERROR TF multiple authority contention:
 * node [/kinect2] publishing transform [kinect2_ir_optical_frame] with parent [kinect2_rgb_optical_frame] already published by node [/cartographer]
 * node [/cartographer] publishing transform [kinect2_ir_optical_frame] with parent [map] already published by node [/kinect2]
```
These errors disappeared after I set `publish_tf` to false in kinect driver. Thanks for your suggestion.

So now I only have these two questions:
* If I add imu sensor, will this improve the 2D map?
* If I move the kinect sensor vertically (change its height in a robot) while running the slam algorithm, will cartographer still be able to generate descent map? If it is fine, does that require imu sensor?

SirVer, 2017-06-01:
> If I add imu sensor, will this improve the 2D map?

Maybe. But it seems unlikely that your map is very bad now and you fix it with just adding IMU. I expect that your setup is poorly tuned.

> If I move the kinect sensor vertically (change its height in a robot) while running the slam algorithm, will cartographer still be able to generate descent map? If it is fine, does that require imu sensor?

No, not if you use 2D SLAM. This violates the 2D assumptions. You could try doing 3D SLAM with your depth camera, but that requires an IMU.

CTTC, 2017-06-01:
Thank you all. @SirVer @Entropy512

closed 2017-06-01 by CTTC

DeepakKarishetti, 2019-02-21:
Sorry for the long time to ask.... Could you tell me are you using this package to build map in real-time, or just record data from kinect and then process the data and build a map?
Thank you for your patience, waiting for your replying..