cartographer-project/cartographer_ros #366: Trajectory visualization crashes

wohe, 2017-06-08:
RViz crashes after a while when trajectory visualization is enabled.

Reading through the RViz code it seems the line strip marker does not support lines with more than 16384 points. See
<https://github.com/ros-visualization/rviz/blob/kinetic-devel/src/rviz/ogre_helpers/billboard_line.cpp>. Maybe this is the issue?

Possible solution: Split longer trajectories into multiple line strip markers and see whether this fixes the crashes.

@ojura Do you want to try a fix? Feel free to unassign  if you would prefer me to look into it.

wohe, 2017-06-08:
@ojura Seems I cannot assign you myself. Please let me know what you think.

ojura, 2017-06-08:
I cannot assign myself either :-) I have reproduced it. I agree, it should be enough to just limit each `visualization_msgs::Marker` to 16384 (16383?) points, and to push as many additional markers as needed into the marker array.

PR #367 (2017-06-09): Fix rviz trajectory rendering crash
Rviz has a limit of 16384 points per marker. To get around this, each
trajectory is split into multiple markers, each up to 16384 points.
Fixes #366.
commit 6658cffa (2017-06-09): Fix rviz trajectory rendering crash (#367)

Rviz has a limit of 16384 points per marker. To get around this, each
trajectory is split into multiple markers, each up to 16384 points.
Fixes #366.
closed 2017-06-09 by wohe