cartographer-project/cartographer_ros #368: Keep spinning while doing the final optimization in the offline node

ojura, 2017-06-08:
This way the effects of the final optimization can also be viewed live in rviz, since everything will still be published (now it's not, because spinning is done only in the bag message processing loop).

Ideas: launch an asyncspinner to handle this. If it's problematic that the time stopped after the bag is finished (e.g. rviz won't redraw the transforms if the time is not moving), the simulation clock could be faked to continue at the same rate as the wall clock when the bag ends.

Also, it could be useful to add an option to keep the offline node running after it saves the assets. Perhaps we still want to look around the map in rviz, and for that we need to keep publishing the submap list.

SirVer, 2017-06-09:
I agree that this would be nice, but your suggested solutions sound quite hackish. I think we  have to acknowledge that ROS does not allow what we'd like here - basically a distinction between data-time and real-time and we have to decide if we are willing to work around or not.

> Also, it could be useful to add an option to keep the offline node running after it saves the assets. Perhaps we still want to look around the map in rviz, and for that we need to keep publishing the submap list.

This I do not like. The quality of the final map should be judged by inspecting the assets, i.e. x-rays and other outputs. The rViz representation is just an approximation really. 

Also, the node can exit, as long as rViz keeps running, it should still display the data, or not? We could change the launch files to remove the `required` keyword on the node, but I do not want to do that because it seems to be the reasonable default for the use case.

ojura, 2017-06-09:
> I agree that this would be nice, but your suggested solutions sound quite hackish.

I know it's hackish, I'm just throwing ideas around :)

> Also, the node can exit, as long as rViz keeps running, it should still display the data, or not? 

That's true, it's still displaying the data but it's easy to disrupt this state. I.e. toggle off/on the visibility of the submaps plugin, or the scan matched point cloud, or the trajectory. These won't get redrawn once they are toggled back on, because they aren't published, since the offline node has quit. If you have a workflow which requires that the map persists in rviz, you're out of luck.

Compare this with the regular online node plus playing a bag - once the bag ends, the node is still alive and publishing everything, listening to services, etc., and you don't have this problem. So I would call it a deficiency of the offline node compared to the offline node.

SirVer, 2017-06-09:
An alternative would be to replay a bag into the online system at 50x times the speed or so. That is similar to the offline system then, but it stays alive. 

You convinced me though that having a `--do-not-exit` or `--keep-running` option in the online node would help.