cartographer-project/cartographer_ros #372: Configuration issues for VLP-16

myu20, 2017-06-09:
Hi,

I am using a VLP-16 indoors and am having trouble tuning the configuration parameters. Currently when driving the robot across a large room the map skews occasionally as seen in the video [here](https://drive.google.com/open?id=0B4qLVF5mv1t3LTVBb0pvenh2cEE) (bag file is in the same folder). My configuration, launch, and urdf files can be found [here](https://github.com/myu20/cartographer_ros).
The velodyne is publishing 1 UDP package per message so the rate is ~750 hz.

I did try plotting the linear acceleration as suggested by this [post](https://github.com/googlecartographer/cartographer_ros/issues/249). The sudden skewing corresponded to spikes in the acceleration, which is probably due to bumps in the floor, but are there any parameters I can change to improve the map?

Thank you!

SirVer, 2017-06-12:
@myu20, you closed the issue. Was your problem resolved? 

To me the video looks like somebody else besides Cartographer is publishing `map -> base_link` and that makes the pose jump. The map looks consistent (no smearing), so Cartographer seems to receive valid data.

myu20, 2017-06-12:
Sorry, I thought I knew how to fix it but I am still having the same problem.
I recorded a new bag file and video (highbay3.mp4 and highbay3.bag in the same folder) in which I drove the robot much slower across the room, but the map rotates whenever the robot rotates.

Nothing else is publishing map -> base_link, so I do not know what is going on.

SirVer, 2017-06-14:
Some random observiations:

- Your velodyne points seem funky. There are coming in with irregular timing (see rqt bag screenshot below) with pretty large gaps.
- Your IMU is only giving you 20Hz - and when you plot the data you can see in the accelerations the bumps you are driving over (as you stated in your OP). They have a large impact on your trajectory, because the IMU data is so rarely sent.
- I was not really able to tune the system around the IMU bumps, sorry. I managed to get reasonable results out of the `highbay3.bag` - by turning on correlative scan matching. That code path is not really mature in 3D though, so not sure if the quality is sufficient. It is also quite slow. See the screenshot below for my result.

My changes are in [this branch](https://github.com/SirVer/cartographer_ros/commits/user/myu20). I suggest you try to up your IMU frequency. Alternatively you could try to detect jumps in the IMU data or the trajectory and start a new trajectory if you see this. Another alternative is to work on the OptimizingTrajectoryBuilder (in code) - it's design promises to be more resilient than the Kalman filter against outliers in data. But it is not yet completed and we did not have time to invest in it for a while.

Let me know if I can help with anything else.

[image]
[image]

closed 2017-06-14 by SirVer

myu20, 2017-06-14:
Ok, thank you so much!

Duke-Allen, 2019-03-02:
@myu20 
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is messy. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Do you know the problem?

artemiialessandrini, 2019-10-07:
Good afternoon, @SirVer , I'm having the same kind of issue.  #1397
Could you please explain or refer how to plot published topics frequencies from rosbag as [you did ](https://user-images.githubusercontent.com/140115/27120677-d1138fea-50e4-11e7-9af6-35d9bdfb419f.png).

I believe it can help me and others to validate rosbag easily
Thank you in advance!

artemiialessandrini, 2019-10-08:
[found it](https://wiki.ros.org/rqt_bag?distro=melodic)
very helpful tool, thank you