cartographer-project/cartographer_ros #379: cartographer_ros build error

myu20, 2017-06-14:
Hi,

I was having issues building Cartographer after git pulling and ended up reinstalling it, but I am now encountering the following error when running `catkin_make_isolated --install --use-ninja --pkg cartographer cartographer_ros ceres-solver`:

`ninja -j6 -l6 in '/home/chaohui/catkin_ws/build_isolated/cartographer_ros'
[15/33] Building CXX object CMakeFiles/cartographer_ros.dir/cartographer_ros/node.cc.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -isystem /usr/include/vtk-5.8 -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /usr/include/eigen3 -isystem /usr/include/ni -isystem /home/chaohui/catkin_ws/install_isolated/include -isystem /home/chaohui/catkin_ws/devel/.private/urdf/include -isystem /home/chaohui/catkin_ws/src/tf2_eigen/include -isystem /home/chaohui/catkin_ws/src/robot_model/urdf/include -isystem /opt/ros/indigo/include -I. -I/home/chaohui/catkin_ws/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12 -Wno-deprecated -O3 -DNDEBUG    -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/node.cc.o -MF CMakeFiles/cartographer_ros.dir/cartographer_ros/node.cc.o.d -o CMakeFiles/cartographer_ros.dir/cartographer_ros/node.cc.o -c /home/chaohui/catkin_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/node.cc
/home/chaohui/catkin_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/node.cc: In member function ‘bool cartographer_ros::Node::HandleFinishTrajectory(cartographer_ros_msgs::FinishTrajectory::Request&, cartographer_ros_msgs::FinishTrajectory::Response&)’:
/home/chaohui/catkin_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/node.cc:456:37: error: ‘cartographer_ros_msgs::FinishTrajectory::Request’ has no member named ‘trajectory_id’
   const int trajectory_id = request.trajectory_id;
`

I found `FinishTrajectory.h` in `/home/chaohui/catkin_ws/install_isolated/include/cartographer_ros_msgs/FinishTrajectory.h`. Any help would be appreciated.

Thank you!

SirVer, 2017-06-14:
Could you try to nuke everything and rebuild? `rm -rf *isolated && catkin_make_isolated`

myu20, 2017-06-14:
Will this delete everything in my workspace?

SirVer, 2017-06-14:
no, this only deletes the build code which will be regenerated by the next build.

myu20, 2017-06-14:
After running `rm -rf *isolated && catkin_make_isolated` I got the same error `cartographer_ros_msgs::FinishTrajectory::Request’ has no member named ‘trajectory_id’ const int trajectory_id = request.trajectory_id`

myu20, 2017-06-14:
I did successfully install and build Cartographer on a different computer, on which I have not installed Cartographer before, without using Ninja.