cartographer-project/cartographer_ros #385: Eigen/Ceres CMake bug causes build error on Zesty

ojura, 2017-06-19:
Hi, I have stumbled into a CMake bug on Zesty when building cartographer_ros. I believe that the bug is in Ceres's `FindEigen.cmake` module. The root of the problem is that finding Eigen, called while finding Ceres, overwrites `PACKAGE_PREFIX_DIR`. To reproduce, execute this script on a fresh Zesty installation (or livecd):

```
ROS_DISTRO=kinetic
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
sudo add-apt-repository universe
sudo apt update
sudo apt install git ros-$ROS_DISTRO-ros-base ninja-build -y 
mkdir -p catkin_ws/src && cd catkin_ws/src
source /opt/ros/$ROS_DISTRO/setup.bash
git clone https://github.com/googlecartographer/cartographer.git
git clone https://ceres-solver.googlesource.com/ceres-solver
git clone https://github.com/googlecartographer/cartographer_ros.git
cd ..
./src/cartographer/scripts/install_debs.sh 
sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y
catkin_make_isolated --use-ninja
```

This fails with:
```
Errors     << cartographer_ros:cmake /home/ubuntu/catkin_ws/logs/cartographer_ros/build.cmake.000.log                                           
CMake Error at /usr/lib/cmake/eigen3/Eigen3Config.cmake:24 (message):
  File or directory /usr/share/cartographer/cmake referenced by variable
  CARTOGRAPHER_CMAKE_DIR does not exist !
Call Stack (most recent call first):
  /home/ubuntu/catkin_ws/devel/share/cartographer/cartographer-config.cmake:57 (set_and_check)
CMakeLists.txt:38 (find_package)
```

Here's a CMake "backtrace":

1. cartographer_ros' CMakeLists.txt calls `find_package(cartographer REQUIRED)`.
2.  In `cartographer-config.cmake`, the command

    ```
    get_filename_component(PACKAGE_PREFIX_DIR "${CMAKE_CURRENT_LIST_DIR}/../../" ABSOLUTE)
    ```
    correctly sets `PACKAGE_PREFIX_DIR` to e.g. `/home/ubuntu/catkin_ws/devel_isolated/cartographer`.
3. In the same file, `find_package(Ceres ...)` is called. Somewhere in the middle of `CeresConfig.cmake`, `find_package(Eigen ...)` is called.
4. Ceres has its own `FindEigen.cmake` module. On Zesty, the following line is reached:

    ```
    find_package(Eigen3 QUIET
                          NO_MODULE
                          NO_CMAKE_PACKAGE_REGISTRY
                          NO_CMAKE_BUILDS_PATH)
    ```
5. This in turn calls `/usr/lib/cmake/eigen3/Eigen3Config.cmake`. On kinetic, this file does almost nothing, but on Zesty, it has some more stuff in it, among it, the following line: `set(PACKAGE_PREFIX_DIR "/usr")`
This overwrites the above variable with an incorrect value which will break building cartographer_ros as described in the next step.
6. Back to  `cartographer-config.cmake`. Finding Ceres is finished, and the following line is executed:

    ```
    set_and_check(CARTOGRAPHER_CMAKE_DIR "${PACKAGE_PREFIX_DIR}/share/cartographer/cmake")
    ```
    This fails, as explained above, because `PACKAGE_PREFIX_DIR` is now wrong.

A quick and dirty mitigation would be to move `set_and_check(CARTOGRAPHER_CMAKE_DIR ...)` before finding Ceres in `cartographer-config.cmake.in` in libcartographer. Pinging @damonkohler, I believe that you're the build system master. :)

SirVer, 2017-06-20:
@sandwichmaker Sameer, are you aware of this problem?

sandwichmaker, 2017-06-21:
@alexmac is the man to look at this.

also even though the summary indicates that this is a problem with ceres, doesn't step 5 indicate that the problem is with Eigen3Config.cmake?

sandwichmaker, 2017-06-21:
@alexsmac is out biking for the next few weeks though.

ojura, 2017-06-21:
Yes, maybe I was too quick to assign blame, I should have pointed out that I am not sure at all whose fault this exactly is. Could be Eigen's, could be Ceres's, could be libcartographer's. I am not familiar enough with usage patterns in CMake in order to make a conclusion.  The best I can is make a pull request in the libcartographer repo with the mitigation I suggested, moving `set_and_check` before finding Ceres.

closed 2017-06-21 by wohe