cartographer-project/cartographer_ros #386: Installation error of cartographer_ros.

jim1949, 2017-06-21:
Hi, this is Jimmy Jiao.
I followed the instruction of notes from http://google-cartographer-ros.readthedocs.io/en/latest/ 
`# Install wstool and rosdep.
sudo apt-get update
sudo apt-get install -y python-wstool python-rosdep ninja-build

# Create a new workspace in 'catkin_ws'.
mkdir catkin_ws
cd catkin_ws
wstool init src

# Merge the cartographer_ros.rosinstall file and fetch code for dependencies.
wstool merge -t src https://raw.githubusercontent.com/googlecartographer/cartographer_ros/master/cartographer_ros.rosinstall
wstool update -t src

# Install deb dependencies.
# The command 'sudo rosdep init' will print an error if you have already
# executed it since installing ROS. This error can be ignored.
sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y

# Build and install.
catkin_make_isolated --install --use-ninja
source install_isolated/setup.bash`

Here when I typed 
`catkin_make_isolated --install --use-ninja`
it tells me a error like this
`
==> Processing catkin package: 'cartographer_ros'
==> Building with env: '/home/jimmy/carto_ws/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/jimmy/carto_ws/build_isolated/cartographer_ros'
ninja: no work to do.
==> ninja -j8 -l8 in '/home/jimmy/carto_ws/build_isolated/cartographer_ros'
[3/10] Building CXX object CMakeFiles/...rtographer_ros/map_builder_bridge.cc.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -Wno-deprecated -O3 -DNDEBUG -isystem /usr/include/vtk-5.8 -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /usr/include/eigen3 -isystem /usr/include/ni -isystem /home/jimmy/carto_ws/install_isolated/include -isystem /opt/ros/indigo/include -I. -I/home/jimmy/carto_ws/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12     -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/map_builder_bridge.cc.o -MF CMakeFiles/cartographer_ros.dir/cartographer_ros/map_builder_bridge.cc.o.d -o CMakeFiles/cartographer_ros.dir/cartographer_ros/map_builder_bridge.cc.o -c /home/jimmy/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/map_builder_bridge.cc
/home/jimmy/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/map_builder_bridge.cc: In member function ‘bool cartographer_ros::MapBuilderBridge::HandleSubmapQuery(cartographer_ros_msgs::SubmapQuery::Request&, cartographer_ros_msgs::SubmapQuery::Response&)’:
/home/jimmy/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/map_builder_bridge.cc:109:38: error: ‘cartographer_ros_msgs::SubmapQuery::Request’ has no member named ‘submap_index’
       request.trajectory_id, request.submap_index, &response_proto);
                                      ^
[3/10] Building CXX object CMakeFiles/...her_ros.dir/cartographer_ros/node.cc.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -Wno-deprecated -O3 -DNDEBUG -isystem /usr/include/vtk-5.8 -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /usr/include/eigen3 -isystem /usr/include/ni -isystem /home/jimmy/carto_ws/install_isolated/include -isystem /opt/ros/indigo/include -I. -I/home/jimmy/carto_ws/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12     -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/node.cc.o -MF CMakeFiles/cartographer_ros.dir/cartographer_ros/node.cc.o.d -o CMakeFiles/cartographer_ros.dir/cartographer_ros/node.cc.o -c /home/jimmy/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/node.cc
/home/jimmy/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/node.cc: In member function ‘bool cartographer_ros::Node::HandleFinishTrajectory(cartographer_ros_msgs::FinishTrajectory::Request&, cartographer_ros_msgs::FinishTrajectory::Response&)’:
/home/jimmy/carto_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/node.cc:456:37: error: ‘cartographer_ros_msgs::FinishTrajectory::Request’ has no member named ‘trajectory_id’
   const int trajectory_id = request.trajectory_id;
                                     ^
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros': 
  Command '['/home/jimmy/carto_ws/install_isolated/env.sh', 'ninja', '-j8', '-l8']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/jimmy/carto_ws/build_isolated/cartographer_ros && /home/jimmy/carto_ws/install_isolated/env.sh ninja -j8 -l8
`
It seems like something is missing... but I don't know how to fix this. Any ideas?
My github repo is: https://github.com/jim1949/carto_ws

SirVer, 2017-06-21:
Can you try going into each directory under src/ and run `git pull` there? It seems like you ran your commands in the short time span where cartographer and cartographer_ros were out of sync.

jim1949, 2017-06-21:
@SirVer Sorry for that, that's because there three .git under these folders in the src/ folder. Thanks for finding this.

closed 2017-06-21 by SirVer