cartographer-project/cartographer_ros #387: ground truth

sunmk2006, 2017-06-21:
1 why does the relation file in [/cartographer/ground_truth/compute_relations_metrics_main.cc](https://github.com/googlecartographer/cartographer/blob/master/cartographer/ground_truth/compute_relations_metrics_main.cc#L129-L130) contain two time stamps,shouldn't the pose of ground truth correspond to only one time?
2 when i use the cartographer_autogenerate_ground_truth tool to generate the relation file of a b3-2016-04-05-15-51-36.bag.pb file which generated from the [b3-2016-04-05-15-51-36.bag](http://google-cartographer-ros.readthedocs.io/en/latest/data.html) file, Why is the output relational file empty?  The output log is shown below：
`./cartographer_autogenerate_ground_truth -pose_graph_filename=/home/vikia/source/data/b3-2016-04-05-15-51-36.bag.pb -output_filename=/home/vikia/source/data/relation.pb
I0621 16:42:05.279050 29829 autogenerate_ground_truth_main.cc:157] Reading pose graph from '/home/vikia/source/data/b3-2016-04-05-15-51-36.bag.pb'...
I0621 16:42:05.279525 29829 autogenerate_ground_truth_main.cc:165] Autogenerating ground truth relations...
I0621 16:42:05.279557 29829 autogenerate_ground_truth_main.cc:92] covered distance size:523
I0621 16:42:05.279568 29829 autogenerate_ground_truth_main.cc:96] submap_to_scan_index size:3
I0621 16:42:05.279573 29829 autogenerate_ground_truth_main.cc:150] Generated 0 relations and ignored 0 outliers.
I0621 16:42:05.279577 29829 autogenerate_ground_truth_main.cc:168] Writing 0 relations to '/home/vikia/source/data/relation.pb'.
`

SirVer, 2017-06-21:
1. No. You want to compare a trajectory locally, therefore you can only meaningfully compare a difference between two poses inside the same trajectory. If the trajectories are similar, you expect these to be similar as well.
2. Probably because the heuristic what makes a good ground truth relation filters all out? I suggest you read through the code, it is quite straightforward.

closed 2017-06-21 by SirVer

sunmk2006, 2017-06-22:
Can you verify that the two code [a](https://github.com/googlecartographer/cartographer/blob/master/cartographer/ground_truth/autogenerate_ground_truth_main.cc#L70-L73) and [b](https://github.com/googlecartographer/cartographer/blob/master/cartographer/ground_truth/autogenerate_ground_truth_main.cc#L104-L107) in the file
is correct and does not cause conflict?

wohe, 2017-06-22:
Can you be more specific, most importantly what did you expect to see in the code and why?