cartographer-project/cartographer_ros #391: Cartographer renders no map with no errors, requires tuning?

MasterEric, 2017-06-22:
When I run Cartographer with my .bag file and configuration, no map renders, but no errors appear in the console. Occasionally,(once every 30 seconds or so), the node will publish a TF frame and the map will render, but this is much less frequently than in the 2d or 3d museum example bags, or even the 2d bag I recorded.

The bag file includes data from 1 Velodyne VLP16, and IMU data with no covariance.

The repository https://github.com/MasterEric/cartographer_ros contains my launch file, urdf, and lua config. I have uploaed my compressed bag file here: https://mega.nz/#!EA9nFBKZ!sZfngNFFJmbkj8beDTNj3MSNNGRcoj05gw4DZoPqsZ4

Additionally I have attached rosout.log.txt for my run of this configuration (it mainly contains rviz errors complaining about cartographer not publishing any tf frames.
[rosout.log.txt](https://github.com/googlecartographer/cartographer_ros/files/1095624/rosout.log.txt)

I would greatly appreciate some help in tuning cartographer given my available data.

SirVer, 2017-06-23:
Looking into the rosout.log.txt, I see:

~~~
1492784455.085094570 INFO [/home/eric/cartographer_second_workspace/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_log_sink.cc:51(ScopedRosLogSink::send) [topics: /rosout, /tf, /submap_list, /trajectory_nodes_list, /scan_matched_points2] I0622 13:10:09.000000 19255 collated_trajectory_builder.cc:84] imu rate: 99.79 Hz 1.00e-02 s +/- 4.39e-04 s (pulsed at 99.39% real time)
1492784455.085094570 INFO [/home/eric/cartographer_second_workspace/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_log_sink.cc:51(ScopedRosLogSink::send) [topics: /rosout, /tf, /submap_list, /trajectory_nodes_list, /scan_matched_points2] I0622 13:10:09.000000 19255 collated_trajectory_builder.cc:84] points2 rate: 9.91 Hz 1.01e-01 s +/- 3.27e-04 s (pulsed at 100.04% real time)
~~~

so the Cartographer node seems happy. Looking through your launch files [I see here](https://github.com/MasterEric/cartographer_ros/blob/master/cartographer_ros/configuration_files/eric_config.lua#L36).

~~~
TRAJECTORY_BUILDER_3D.scans_per_accumulation = 160
~~~

My guess is that this is wrong for your data and you probably want to set it to 1. See [this faq entry](https://google-cartographer-ros.readthedocs.io/en/latest/faq.html#the-laser-data-in-the-3d-bags-is-much-higher-than-the-maximum-reported-20-hz-rotation-speed-that-the-vlp-16-can-do-why).

closed 2017-06-23 by SirVer

MasterEric, 2017-06-26:
This was exactly the change I needed to get this working! Thank you so much! :grin:

SirVer, 2017-06-27:
Seems like it did not loop close correctly on the right. I see a duplicated wall.